brocaar/chirpstack-network-server #323: Index out of range error when transmitting MAC commands to an ABP device

amrbekhit, 2018-05-16:
I'm trying to pair a TBS-220 park sensor to the loraserver. It's an ABP device. We've created a new device profile for ABP-devices and then created the application and device entry for this device. Our gateways are using the standard Semtech packet forwarder running behind the lora-gateway-bridge.

As soon as the ABP device is turned on, loraserver crashes with the following error:

```
INFO[0007] rx info sent to network-controller            dev_eui=70b3d53af0030385
INFO[0007] backend/gateway: rx packet received
ERRO[0007] finished client unary call                    error="rpc error: code = Internal desc = get device error: rpc error: code = Unavailable desc = all SubConns are in TransientFailure" grpc.code=Internal grpc.method=HandleUplinkData grpc.service=as.ApplicationServer grpc.time_ms=5.504 span.kind=client system=grpc
ERRO[0007] publish uplink data to application-server error  error="rpc error: code = Internal desc = get device error: rpc error: code = Unavailable desc = all SubConns are in TransientFailure"
INFO[0007] device-session saved                          dev_addr=70030385 dev_eui=70b3d53af0030385
INFO[0007] adr request added to mac-command queue        dev_eui=70b3d53af0030385 dr=0 nb_trans=1 req_dr=5 req_nb_trans=1 req_tx_power_idx=1 tx_power=0
INFO[0007] requesting device-status                      dev_eui=70b3d53af0030385
panic: runtime error: index out of range

goroutine 45 [running]:
github.com/brocaar/loraserver/internal/downlink/data.setMACCommands.func1(0xc4200a3b00, 0x0, 0x0)
        /home/travis/gopath/src/github.com/brocaar/loraserver/internal/downlink/data/data.go:446 +0x77c
github.com/brocaar/loraserver/internal/downlink/data.HandleResponse(0x4, 0xc31fa0, 0xc420095a90, 0xf97973e1, 0x33c134e0, 0x0, 0xc420034f0c, 0x4, 0xc, 0x7d, ...)
        /home/travis/gopath/src/github.com/brocaar/loraserver/internal/downlink/data/data.go:190 +0x1d3
github.com/brocaar/loraserver/internal/uplink/data.handleDownlink(0xc4200a2d80, 0x0, 0x0)
        /home/travis/gopath/src/github.com/brocaar/loraserver/internal/uplink/data/data.go:426 +0x194
github.com/brocaar/loraserver/internal/uplink/data.Handle(0x4, 0xc31fa0, 0xc420095a90, 0xf97973e1, 0x33c134e0, 0x0, 0xc420034f0c, 0x4, 0xc, 0x7d, ...)
        /home/travis/gopath/src/github.com/brocaar/loraserver/internal/uplink/data/data.go:65 +0x110
github.com/brocaar/loraserver/internal/uplink.collectPackets.func1(0x4, 0xc31fa0, 0xc420095a90, 0xf97973e1, 0x33c134e0, 0x0, 0xc420034f0c, 0x4, 0xc, 0x7d, ...)
        /home/travis/gopath/src/github.com/brocaar/loraserver/internal/uplink/uplink.go:80 +0xa6
github.com/brocaar/loraserver/internal/uplink.collectAndCallOnce(0xc4200f5780, 0x201003492074652, 0x0, 0x0, 0x893965e4, 0x33c134e0, 0x6, 0x1, 0x1, 0xc420530fcb, ...)
        /home/travis/gopath/src/github.com/brocaar/loraserver/internal/uplink/collect.go:119 +0x10a5
github.com/brocaar/loraserver/internal/uplink.collectPackets(0x201003492074652, 0x0, 0x0, 0x893965e4, 0x33c134e0, 0x6, 0x1, 0x1, 0xc420530fcb, 0x3, ...)
        /home/travis/gopath/src/github.com/brocaar/loraserver/internal/uplink/uplink.go:71 +0x7b
github.com/brocaar/loraserver/internal/uplink.HandleRXPacket(0x201003492074652, 0x0, 0x0, 0x893965e4, 0x33c134e0, 0x6, 0x1, 0x1, 0xc420530fcb, 0x3, ...)
        /home/travis/gopath/src/github.com/brocaar/loraserver/internal/uplink/uplink.go:67 +0x5c
github.com/brocaar/loraserver/internal/uplink.HandleRXPackets.func1(0xc420269ec0, 0x201003492074652, 0x0, 0x0, 0x893965e4, 0x33c134e0, 0x6, 0x1, 0x1, 0xc420530fcb, ...)
        /home/travis/gopath/src/github.com/brocaar/loraserver/internal/uplink/uplink.go:57 +0xaa
created by github.com/brocaar/loraserver/internal/uplink.HandleRXPackets
        /home/travis/gopath/src/github.com/brocaar/loraserver/internal/uplink/uplink.go:54 +0x135
```

brocaar, 2018-05-16:
Which version are you using?

amrbekhit, 2018-05-16:
Sorry forgot to mention:

loraserver 0.26.2
lora-app-server 0.20.2
lora-gateway-bridge 2.4.0

brocaar, 2018-05-17:
Thanks for reporting and proposing a fix! I've implemented it in a slightly different way so that it became easier to test only this function + I've added some testcases which would also replicate this issue to make sure that this won't happen again in the future :-)

Sorry about this :-( I will directly issue a new bugfix release!

amrbekhit, 2018-05-17:
Glad to have contributed - I specifically learnt Go to find and fix this bug. As we're hoping to use loraserver more and more, I hope to be able to contribute more.

Thanks for all your work writing loraserver! Great piece of software.