brocaar/chirpstack-network-server #326: Join requests ignored after third successful join

bldh, 2018-05-30:
### Is this a bug or a feature request?
Bug
### What did you expect?
Join request to be accepted.
### What happened?
After the device has joined three times not all future join requests are accepted.
### What version are your using?
0.26.3
### How can your issue be reproduced?
Repeatedly join a device to a network.

[image]

brocaar, 2018-06-05:
Have you looked at the LoRa Server log output?

bldh, 2018-06-18:
> Jun 18 16:08:25 loraserver.local loraserver[1223]: time="2018-06-18T16:08:25+10:00" level=info msg="packet(s) collected" dev_eui=deadc0de00000006 gw_count=1 gw_macs=98208edef4600103 mtype=JoinRequest
Jun 18 16:08:25 loraserver.local loraserver[1223]: time="2018-06-18T16:08:25+10:00" level=error msg="processing rx packet error: get device error: object does not exist" data_base64="AAAAAAAAAAAABgAAAN7Ard4y4WeU9Dk="
Jun 18 16:08:34 loraserver.local loraserver[1223]: time="2018-06-18T16:08:34+10:00" level=info msg="backend/gateway: rx packet received"
Jun 18 16:08:34 loraserver.local loraserver[1223]: time="2018-06-18T16:08:34+10:00" level=info msg="packet(s) collected" dev_eui=deadc0de00000006 gw_count=1 gw_macs=98208edef4600103 mtype=JoinRequest
Jun 18 16:08:34 loraserver.local loraserver[1223]: time="2018-06-18T16:08:34+10:00" level=error msg="processing rx packet error: get device error: object does not exist" data_base64="AAAAAAAAAAAABgAAAN7Ard4oeqiwXn4="
Jun 18 16:08:40 loraserver.local loraserver[1223]: time="2018-06-18T16:08:40+10:00" level=info msg="backend/gateway: rx packet received"
Jun 18 16:08:40 loraserver.local loraserver[1223]: time="2018-06-18T16:08:40+10:00" level=info msg="packet(s) collected" dev_eui=deadc0de00000006 gw_count=1 gw_macs=98208edef4600103 mtype=JoinRequest
Jun 18 16:08:40 loraserver.local loraserver[1223]: time="2018-06-18T16:08:40+10:00" level=error msg="processing rx packet error: get device error: object does not exist" data_base64="AAAAAAAAAAAABgAAAN7Ard6hQgOpOOg="
Jun 18 16:08:42 loraserver.local loraserver[1223]: time="2018-06-18T16:08:42+10:00" level=info msg="backend/gateway: gateway stats packet received" mac=98208edef4600103

This is all that appears in the logs while rejoining a device.

brocaar, 2018-06-18:
> Jun 18 16:08:40 loraserver.local loraserver[1223]: time="2018-06-18T16:08:40+10:00" level=error msg="processing rx packet error: get device error: object does not exist"

bldh, 2018-06-21:
The device exists within a valid application on loraserver instance and has joined previously. After the server accepts the join requests the device functions as expected with no packets being ignored, so I'm not quite sure as to why the object would not exist when the device tries to rejoin.

bldh, 2018-07-19:
Any update on this?

bldh, 2018-08-27:
Hi, this is still an issue.

Currently running v2.0.1 of lora app server, v2.0.2 of lora server. As previously stated the devices are created and after a few unanswered join requests lora server sends a join accept.

Can I please get an update on this,

Regards,  Brendan.

brocaar, 2018-10-16:
> Any update on this?

I'm not able to reproduce and it hasn't been reported by others. I don't think this is a LoRa Server issue.

When you get the error:

> Jun 18 16:08:40 loraserver.local loraserver[1223]: time="2018-06-18T16:08:40+10:00" level=error msg="processing rx packet error: get device error: object does not exist" data_base64="AAAAAAAAAAAABgAAAN7Ard6hQgOpOOg="

Then it indicates that the device could not be retrieved from the database (eg. using the DevEUI from the join-request).

closed 2018-10-16 by brocaar

Brulath, 2019-02-02:
Hey brocaar,

I've determined the way to cause this is to run two instances of the network server (AU915 and AS923) sharing the same MQTT server/gateway bridge/redis (different postgres databases). Whenever both servers are operating one or the other will randomly fail to respond to incoming packets.

As far as I can tell one server will consume messages intended for the other. We have gateways configured with a service profile for the AU915 server, with a device profile also configured for AU915, but those messages are randomly consumed by the AS923 server (with errors about "channel does not exist").

I assume some more detailed configuration is required to run multiple network servers successfully. Is that documented somewhere?

Thanks for your time :)

- B

brocaar, 2019-02-11:
Each environment should use a different MQTT broker (or topic prefix) and an isolated Redis database. When you share these, it is very likely that you run into issues like these.

kees4course, 2019-02-21:
I am experiencing the same problem: it takes some (re)joinrequest before the joinaccept is send. When removing the device from loraserver and then adding it again, no delays immediate response...
I have only one process running of each on a Rpi (so no two instances of a the networkserver)

kees4course, 2019-02-22:
Other reason for refusing a re-join is the DevNonce, which should never being used before... Above is solved.