brocaar/chirpstack-network-server #327: Increased gateway information and reporting

danielknox, 2018-06-01:
Hi, this is a feature request.

It would be useful if gateways that have been registered to an organisation, via the app-server, can raise an alert (email, etc.) if they have not been seen for a user definable length of time. Similarly increased reporting, to help identify misuse from other organisations nodes, would be useful - this detail could include common spread factors, frequencies and top nodes within certain periods.

This will help improve and encourage gateway management via the network server.

chopmann, 2019-09-04:
@brocaar maybe use the prometheus endpoint. The rest should be alert manager stuff.

brocaar, 2020-10-05:
I'm closing this issue. In the last releases there have been some improvements (e.g. org dashboard). I'm not planning to implement alerting etc. into ChirpStack. I think other tools (Grafana, Prometheus, ...) are better at this job as they provide a wide range of integrations (email, pushover, pagerduty, ...).

closed 2020-10-05 by brocaar