brocaar/chirpstack-network-server #328: Automigration not happening afgter fresh installation

urbie-mk2, 2018-06-03:
### Is this a bug or a feature request?
seeking help / error reporting ?
### What did you expect?
database migration
### What happened?
no schema created
### What version are your using?
go 1.10.2
loraserver 1.0 (but happened before with earlier versions)
psql 10.4
### How can your issue be reproduced?
drop all tables, build loraserver, create config, and run loraserver (./build/loraserver)
### Could you share your log output?

Database content after running loraserver once:
```
loraserver_ns=# \d
                List of relations
 Schema |      Name       | Type  |     Owner     
--------+-----------------+-------+---------------
 public | gorp_migrations | table | loraserver_ns
(1 row)

loraserver_ns=# select * from gorp_migrations 
loraserver_ns-# ;
 id | applied_at 
----+------------
(0 rows)
```

config file:
```
[general]
log_level=4
[postgresql]
dsn="postgres://loraserver_ns:dbpassword@localhost/loraserver_ns?sslmode=disable"
automigrate=true
...
```

Log after start of loraserver:
```
INFO[0000] starting LoRa Server                          band=EU_863_870 docs="https://docs.loraserver.io/" net_id=000000 version=1.0.0-1-g3338792
INFO[0000] setup redis connection pool                   url="redis://localhost:6379"
INFO[0000] connecting to postgresql                     
INFO[0000] backend/gateway: TLS config is empty         
INFO[0000] backend/gateway: connecting to mqtt broker    server="tcp://localhost:1883"
INFO[0000] configuring join-server client                ca_cert= server="http://localhost:8003" tls_cert= tls_key=
INFO[0000] no network-controller configured             
INFO[0000] applying database migrations                 
INFO[0000] backend/gateway: connected to mqtt server    
INFO[0000] backend/gateway: subscribing to rx topic      qos=0 topic=gateway/+/rx                            
INFO[0000] backend/gateway: subscribing to stats topic   qos=0 topic=gateway/+/stats
INFO[0000] migrations applied                            count=0
INFO[0000] starting api server                           bind="0.0.0.0:8000" ca-cert= tls-cert= tls-key=
INFO[0000] starting downlink device-queue scheduler     
ERRO[0000] class-c scheduler error                       error="get deveuis with class-c device-queue items error: select error: pq: relation \"device\" does not exist"
ERRO[0001] class-c scheduler error                       error="get deveuis with class-c device-queue items error: select error: pq: relation \"device\" does not exist"
...
```

`No migration was done *count = 0)` ( LoRaServer  does run the migration funciton)

More Info:
- `github.com/rubenv/sql-migrate/migrate.go`: `(func (a AssetMigrationSource) FindMigrations()` does not find any assets:  `a.AssetDir(a.Dir)` returns empty slice.
- (`AssetDir` is  defined in the `migrations_gen.go` file) (`node.Children == 0` / `_bintree.Children == 0`, `node.Func == 0`). I suppose bintree should contain the sql as embedded binary but I cannot explain why it isnt there.

go-bindata does generate the migrations_gen.go (triggered in main.go)
[migrations_gen.zip](https://github.com/brocaar/loraserver/files/2065942/migrations_gen.zip)

PS: While having a look at go-bindata I found this note: https://github.com/jteeuwen/go-bindata/issues/5 indicating that the repository is not updated anymore.

brocaar, 2018-06-04:
Not sure what is going wrong at your side, but using the latest pre-compiled build, I don't have any issues. The commands executed by the CI platform can be found here: https://github.com/brocaar/loraserver/blob/master/.travis.yml.

urbie-mk2, 2018-06-08:
maybe it is postgres, I use version 10. I will check again on the weekend.

closed 2018-08-10 by brocaar