brocaar/chirpstack-network-server #333: LoRa Server Network Server cannot start via systemctl

adamchance365, 2018-06-19:
### Is this a bug or a feature request?
Bug

### What did you expect?
LoRa Server Network Server to start normally via systemctl

### What happened?
LoRa Server Network Server failed with the `Jun 19 10:56:14 ac-lorawan-server-test loraserver[8843]: time="2018-06-19T10:56:14+01:00" level=error msg="ping database error, will retry in 2s: stat /bin/false/.postgresql/postgresql.crt: not a directory` error

### What version are your using?
1.0.0

### How can your issue be reproduced?
`systemctl start loraserver`

### Could you share your log output?
`Jun 19 10:55:22 ac-lorawan-server-test systemd[1]: Started LoRa Server.
Jun 19 10:55:22 ac-lorawan-server-test loraserver[8843]: time="2018-06-19T10:55:22+01:00" level=info msg="starting LoRa Server" band=EU_863_870 docs="https://docs.loraserver.io/" net_id=000
Jun 19 10:55:22 ac-lorawan-server-test loraserver[8843]: time="2018-06-19T10:55:22+01:00" level=info msg="setup redis connection pool" url="redis://localhost:6379"
Jun 19 10:55:22 ac-lorawan-server-test loraserver[8843]: time="2018-06-19T10:55:22+01:00" level=info msg="connecting to postgresql"
Jun 19 10:56:14 ac-lorawan-server-test loraserver[8843]: time="2018-06-19T10:56:14+01:00" level=error msg="ping database error, will retry in 2s: stat /bin/false/.postgresql/postgresql.crt: not a directory`

adamchance365, 2018-06-19:
This issue can be resolved by changing the `home` directory of the `loraserver` user from `/bin/false` to `/etc/loraserver/`

brocaar, 2018-06-25:
I have never seen this issue myself and I'm using ubuntu 16.04 too. Are you connecting with PostgreSQL using TLS?

adamchance365, 2018-06-27:
We are connecting to a local PostgreSQL server using peer authentication

commit c1bd09ea (2018-10-02): Set homedir to /etc/loraserver (deb installer).

Closes #333.
closed 2018-10-02 by brocaar