brocaar/chirpstack-network-server #338: Install LORA SERVER LORA APP SERVER LORA GATEWAY in Windows 7 64 Bit

sahanaRavindran, 2018-07-06:
### Is this a bug or a feature request?
I am new to LORA. Am using Windows 64 bit in my system.I have installed MQTT,Redis database,PostgreSQL. 
### What did you expect?
Help to install LORA server Lora app server and lora gateway in windows 7.
there is no configfile too..how to install that??
### What happened?
While installing loraserver using exe file in cmd prompt.
### What version are your using?
LORASERVER version   loraserver_1.0.0_windows_amd64
### How can your issue be reproduced?
Kindly help me to install loraserver loraappserver lora gateway.
### Could you share your log output?
[image]

brocaar, 2018-07-06:
I think the error explains it all. Configuration examples are available on www.loraserver.io.

closed 2018-07-06 by brocaar

sahanaRavindran, 2018-07-06:
I have installed the following before installing LORA SERVER,LORA APP SERVER AND LORA GATEWAY ,

1.MQTT version 1.4.15
2.Redis version 64.3.0.503
3.PostgreSQL version 9.6

I have downloaded loraserver.exe while running exe in cmd prompt it is showing ping database error..will retry in 2s...
For configuration , i have referred this link https://www.loraserver.io/install/configuration/ and got .toml fileis created in my current working directory.
am not aware of .toml..what to do with that??
Still the above mentioned error is persisting...
kindly explain me..