brocaar/chirpstack-network-server #340: Use of the board and antenna properties

amrbekhit, 2018-07-27:
A while ago in order to add support for the Kerlink iBTS, the board/brd and antenna/ant properties were added to the JSON packets moving between the gateways and the server. Other than the application and gateway specifying the board and antenna values, does the loraserver actually do any processing with these values?

The reason I ask is because we're planning to have a setup whereby several individual packet forwarders sit behind piece of software that manages their channels, but appears as a single gateway to the loraserver. In order for this to work, when a downlink comes from the loraserver in response to a previous uplink, we need to know which board sent the original uplink so that the downlink can be routed correctly. When I saw the board and antenna properties, for some reason I assumed that the loraserver would set the downlink board and antenna properties to match the uplink, but this doesn't appear to be the case. For example, if I send a confirmed uplink with board=1, the downlink ack always has board set to 0.

This is something which I could potentially work towards adding into the loraserver, but I don't know if it will conflict with the original iBTS usage that the board and antenna parameters were originally added for.

brocaar, 2018-07-30:
I think this is a bug and should be handled as you describe, board=1 and antenna=2 uplink means a downlink with the same board / antenna properties.

amrbekhit, 2018-07-30:
Thanks for your reply - I'm happy to work towards implementing this and submitting a pull request. I have one question: you mention it is a bug, which implies that the functionality has been implemented but for some reason is not working as expected. My quick search through the code seems to indicate that such functionality is not present at all in the code. Can you confirm whether this is the case or not, please?

brocaar, 2018-07-30:
With bug I mean that I did implement the uplink part, but as you found out, the downlink part is not mapped. I believe that this should only be a couple lines of code, as all information should be available for downlink :-) 

If you would like to create a pull-request, great!

(note that I do have an iBTS for testing now!)

Could you share something about your project on https://forum.loraserver.io/ (multiple gateways behind one piece of software to bundle their data as a single gateway)? Sounds interesting!

closed 2018-08-18 by brocaar