brocaar/chirpstack-network-server #344: The downlink can't be send in the Class-B mode.

andrewrene, 2018-08-07:
### Is this a bug or a feature request?

Bug

### What did you expect?

In the Class-B mode, the ping-solt downlink should be sent from the forwarder side.

### What version are your using?

ISM Band : EU868
Beacon Packet Forwarder for Lora Gateway :  4.0.1
Lora concentrator HAL library version info : 5.0.1
End-Node : LoRaWAN1.1
LoRa Gateway Bridge : 2.4.1
GNSS module : GMM-G3

### Could you share your log output?

I tested the class-b function but alway get the jit error during downlink, the value in the tmms field from JSON down message seems wrong so the ping-solt downlink is always rejected by forwarder side,  the  logs as below,  

*. On the end-node side log

Network Joined
~DEVICE_STATE_REQ_BEACON_TIMING
Frame send
~DEVICE_STATE_BEACON_ACQUISITION
MLME_BEACON_ACQUISITION, Received Acquisition.
MLME_BEACON, Received BEACON:1
~DEVICE_STATE_REQ_PINGSLOT_ACK
Frame send
~DEVICE_STATE_REQ_PINGSLOT_ACK
Frame send
MLME_PING_SLOT_INFO, CLASS_B
Frame send
Frame send
Frame send
Frame send
MLME_BEACON, Received BEACON:2
….

*. On the forwarder side log

….
JSON down: {"txpk":{"imme":false,"tmms":1217653456000,"freq":869.525,"rfch":0,"powe":14,"modu":"LORA","datr":"SF12BW125","codr":"4/5","ipol":true,"size":23,"data":"oAys0wGFBAADVwcAAQozTmQZLi4QsYM=","brd":0,"ant":0}}
INFO: [down] a packet will be sent on timestamp value 344686124 (calculated from GPS time)
src/jitqueue.c:140:jit_enqueue(): Current concentrator time is 346122413, pkt_type=1
src/jitqueue.c:251:jit_enqueue(): ERROR: Packet REJECTED, timestamp seems wrong, too much in advance (current=346122413, packet=344686124, type=1)
ERROR: Packet REJECTED (jit error=2)
……

*. The downlink command 
mosquitto_pub -h 192.168.88.136 -t application/1/device/2000000000000021/tx -m "{\"reference\": \"abcd1234\",\"confirmed\":true,\"fPort\":10,\"data\":\"OTQ1Mwo=\"}" -q 1 -u admin -P admin

*. The device profile
[image]

brocaar, 2018-08-10:
Let me validate this. I'll come back to this next week.

brocaar, 2018-08-16:
I can't reproduce the issue and all is working fine here (tested with Multitech Conduit with onboard GNSS module). I don't think your GNSS module is supported for Class-B:

https://github.com/Lora-net/lora_gateway/blob/a955619271b5d0a46d32e08150acfbc1eed183b7/libloragw/src/loragw_gps.c#L281

closed 2018-08-16 by brocaar