brocaar/chirpstack-network-server #345: delete device-profiles from cache error: ERR wrong number of arguments for 'del' command

saboohajaz, 2018-08-15:
Can't start Lora server

pi@raspberrypi:~ $ sudo loraserver
INFO[0000] starting LoRa Server                          band=AS_923 docs="https://docs.loraserver.io/" net_id=000000 version=2.0.1
INFO[0000] disabling all channels
INFO[0000] enabling channels                             channels="[0 1 2 3 4 5 6]"
INFO[0000] setup redis connection pool                   url="redis://localhost:6379"
INFO[0000] connecting to postgresql
INFO[0000] backend/gateway: TLS config is empty
INFO[0000] backend/gateway: connecting to mqtt broker    server="tcp://localhost:1883"
INFO[0000] configuring join-server client                ca_cert= server="http://localhost:8003" tls_cert= tls_key=
INFO[0000] no network-controller configured
INFO[0000] applying database migrations
INFO[0000] backend/gateway: connected to mqtt server
INFO[0000] backend/gateway: subscribing to rx topic      qos=0 topic=gateway/+/rx
INFO[0000] backend/gateway: subscribing to stats topic   qos=0 topic=gateway/+/stats
INFO[0000] migrations applied                            count=0
DEBU[0000] sql query executed                            args="[]" duration="959.005µs" query="lock table code_migration"
DEBU[0000] sql query executed                            args="[v1_to_v2_flush_profiles_cache 2018-08-15 06:46:57.211733462 +0000 UTC m=+0.103878003]" duration=3.132639ms query="\n\t\t\tinsert into code_migration (\n\t\t\t\tid,\n\t\t\t\tapplied_at\n\t\t\t) values ($1, $2)\n\t\t\ton conflict\n\t\t\t\tdo nothing\n\t\t"
DEBU[0000] sql query executed                            args="[]" duration=10.943896ms query="\n\t\tselect\n\t\t\tdevice_profile_id\n\t\tfrom\n\t\t\tdevice_profile\n\t"
FATA[0000] delete device-profiles from cache error: ERR wrong number of arguments for 'del' command

--------------------------------------------------------------------------------
pi@raspberrypi:~ $ journalctl -f -n 30 -u loraserver
-- Logs begin at Thu 2016-11-03 17:16:43 UTC. --
Aug 15 06:40:52 raspberrypi systemd[1]: loraserver.service: Failed with result 'exit-code'.
Aug 15 06:40:52 raspberrypi systemd[1]: loraserver.service: Service hold-off time over, scheduling restart.
Aug 15 06:40:52 raspberrypi systemd[1]: Stopped LoRa Server.
Aug 15 06:40:52 raspberrypi systemd[1]: Started LoRa Server.
Aug 15 06:40:52 raspberrypi loraserver[1602]: time="2018-08-15T06:40:52Z" level=info msg="starting LoRa Server" band=AS_923 docs="https://docs.loraserver.io/" net_id=000000 version=2.0.1
Aug 15 06:40:52 raspberrypi loraserver[1602]: time="2018-08-15T06:40:52Z" level=info msg="disabling all channels"
Aug 15 06:40:52 raspberrypi loraserver[1602]: time="2018-08-15T06:40:52Z" level=info msg="enabling channels" channels="[0 1 2 3 4 5 6]"
Aug 15 06:40:52 raspberrypi loraserver[1602]: time="2018-08-15T06:40:52Z" level=info msg="setup redis connection pool" url="redis://localhost:6379"
Aug 15 06:40:52 raspberrypi loraserver[1602]: time="2018-08-15T06:40:52Z" level=info msg="connecting to postgresql"
Aug 15 06:40:52 raspberrypi loraserver[1602]: time="2018-08-15T06:40:52Z" level=info msg="backend/gateway: TLS config is empty"
Aug 15 06:40:52 raspberrypi loraserver[1602]: time="2018-08-15T06:40:52Z" level=info msg="backend/gateway: connecting to mqtt broker" server="tcp://localhost:1883"
Aug 15 06:40:52 raspberrypi loraserver[1602]: time="2018-08-15T06:40:52Z" level=info msg="backend/gateway: connected to mqtt server"
Aug 15 06:40:52 raspberrypi loraserver[1602]: time="2018-08-15T06:40:52Z" level=info msg="backend/gateway: subscribing to rx topic" qos=0 topic=gateway/+/rx
Aug 15 06:40:52 raspberrypi loraserver[1602]: time="2018-08-15T06:40:52Z" level=info msg="configuring join-server client" ca_cert= server="http://localhost:8003" tls_cert= tls_key=
Aug 15 06:40:52 raspberrypi loraserver[1602]: time="2018-08-15T06:40:52Z" level=info msg="no network-controller configured"
Aug 15 06:40:52 raspberrypi loraserver[1602]: time="2018-08-15T06:40:52Z" level=info msg="applying database migrations"
Aug 15 06:40:52 raspberrypi loraserver[1602]: time="2018-08-15T06:40:52Z" level=info msg="backend/gateway: subscribing to stats topic" qos=0 topic=gateway/+/stats
Aug 15 06:40:52 raspberrypi loraserver[1602]: time="2018-08-15T06:40:52Z" level=info msg="migrations applied" count=0
Aug 15 06:40:52 raspberrypi loraserver[1602]: time="2018-08-15T06:40:52Z" level=debug msg="sql query executed" args="[]" duration="656.608µs" query="lock table code_migration"
Aug 15 06:40:52 raspberrypi loraserver[1602]: time="2018-08-15T06:40:52Z" level=debug msg="sql query executed" args="[v1_to_v2_flush_profiles_cache 2018-08-15 06:40:52.604238441 +0000 UTC m=+0.061488914]" duration=1.982791ms query="\n\t\t\tinsert into code_migration (\n\t\t\t\tid,\n\t\t\t\tapplied_at\n\t\t\t) values ($1, $2)\n\t\t\ton conflict\n\t\t\t\tdo nothing\n\t\t"
Aug 15 06:40:52 raspberrypi loraserver[1602]: time="2018-08-15T06:40:52Z" level=debug msg="sql query executed" args="[]" duration=11.586071ms query="\n\t\tselect\n\t\t\tdevice_profile_id\n\t\tfrom\n\t\t\tdevice_profile\n\t"
Aug 15 06:40:52 raspberrypi systemd[1]: loraserver.service: Main process exited, code=exited, status=1/FAILURE
Aug 15 06:40:52 raspberrypi systemd[1]: loraserver.service: Unit entered failed state.
Aug 15 06:40:52 raspberrypi systemd[1]: loraserver.service: Failed with result 'exit-code'.
Aug 15 06:40:52 raspberrypi systemd[1]: loraserver.service: Service hold-off time over, scheduling restart.
Aug 15 06:40:52 raspberrypi systemd[1]: Stopped LoRa Server.
Aug 15 06:40:52 raspberrypi systemd[1]: loraserver.service: Start request repeated too quickly.
Aug 15 06:40:52 raspberrypi systemd[1]: Failed to start LoRa Server.
Aug 15 06:40:52 raspberrypi systemd[1]: loraserver.service: Unit entered failed state.
Aug 15 06:40:52 raspberrypi systemd[1]: loraserver.service: Failed with result 'exit-code'.

saboohajaz, 2018-08-15:
Application server LOG
Aug 15 07:13:59 raspberrypi lora-app-server[1806]: time="2018-08-15T07:13:59Z" level=warning msg="creating insecure network-server client" server="localhost:8000"
Aug 15 07:13:59 raspberrypi lora-app-server[1806]: time="2018-08-15T07:13:59Z" level=warning msg="grpc: addrConn.createTransport failed to connect to {localhost:8000 0  <nil>}. Err :connection error: desc = \"transport: Error while dialing dial tcp [::1]:8000: connect: connection refused\". Reconnecting..."
Aug 15 07:13:59 raspberrypi lora-app-server[1806]: time="2018-08-15T07:13:59Z" level=warning msg="Failed to dial localhost:8000: context canceled; please retry."
Aug 15 07:13:59 raspberrypi lora-app-server[1806]: time="2018-08-15T07:13:59Z" level=error msg="finished unary call with code Unknown" error="rpc error: code = Unknown desc = context deadline exceeded" grpc.code=Unknown grpc.method=Create grpc.service=api.NetworkServerService grpc.start_time="2018-08-15T07:13:58Z" grpc.time_ms=529.087 peer.address="[::1]:43306" span.kind=server system=grpc
Aug 15 07:14:47 raspberrypi lora-app-server[1806]: time="2018-08-15T07:14:47Z" level=warning msg="creating insecure network-server client" server="localhost:8000"
Aug 15 07:14:47 raspberrypi lora-app-server[1806]: time="2018-08-15T07:14:47Z" level=warning msg="grpc: addrConn.createTransport failed to connect to {localhost:8000 0  <nil>}. Err :connection error: desc = \"transport: Error while dialing dial tcp [::1]:8000: connect: connection refused\". Reconnecting..."
Aug 15 07:14:48 raspberrypi lora-app-server[1806]: time="2018-08-15T07:14:48Z" level=warning msg="Failed to dial localhost:8000: context canceled; please retry."
Aug 15 07:14:48 raspberrypi lora-app-server[1806]: time="2018-08-15T07:14:48Z" level=error msg="finished unary call with code Unknown" error="rpc error: code = Unknown desc = context deadline exceeded" grpc.code=Unknown grpc.method=Create grpc.service=api.NetworkServerService grpc.start_time="2018-08-15T07:14:47Z" grpc.time_ms=527.037 peer.address="[::1]:43306" span.kind=server system=grpc
Aug 15 07:14:52 raspberrypi lora-app-server[1806]: time="2018-08-15T07:14:52Z" level=warning msg="creating insecure network-server client" server="localhost:8000"
Aug 15 07:14:52 raspberrypi lora-app-server[1806]: time="2018-08-15T07:14:52Z" level=warning msg="grpc: addrConn.createTransport failed to connect to {localhost:8000 0  <nil>}. Err :connection error: desc = \"transport: Error while dialing dial tcp [::1]:8000: connect: connection refused\". Reconnecting..."
Aug 15 07:14:52 raspberrypi lora-app-server[1806]: time="2018-08-15T07:14:52Z" level=warning msg="Failed to dial localhost:8000: context canceled; please retry."
Aug 15 07:14:52 raspberrypi lora-app-server[1806]: time="2018-08-15T07:14:52Z" level=error msg="finished unary call with code Unknown" error="rpc error: code = Unknown desc = context deadline exceeded" grpc.code=Unknown grpc.method=Create grpc.service=api.NetworkServerService grpc.start_time="2018-08-15T07:14:52Z" grpc.time_ms=526.207 peer.address="[::1]:43306" span.kind=server system=grpc

closed 2018-08-15 by brocaar

brocaar, 2018-08-15:
Cheers, this has been fixed in v2.0.2. Sorry for this, I completely missed this testcase 🤦‍♂️

commit 3a2d7198 (2018-08-15): Fix flush profiles cache on empty db.

Closes #345.