brocaar/chirpstack-network-server #349: runtime error: invalid memory address or nil pointer dereference

sleepingAnt, 2018-08-20:
Hi, every time I received rx packet, I get errors below, and loraserver crashed.

```
INFO[0103] backend/gateway: rx packet received
INFO[0103] creating application-server client            server="localhost:8001"
INFO[0103] rx info sent to network-controller            dev_eui=401f000010ffffff
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x30 pc=0xa19886]

goroutine 285 [running]:
github.com/brocaar/loraserver/internal/uplink/data.sendFRMPayloadToApplicationServer.func1(0x0, 0x0, 0xc4200c09cc, 0x8, 0x8, 0xc4200c09d4, 0x8, 0x8, 0x2100000010, 0x500000001, ...)
        /home/travis/gopath/src/github.com/brocaar/loraserver/internal/uplink/data/data.go:403 +0xc6
created by github.com/brocaar/loraserver/internal/uplink/data.sendFRMPayloadToApplicationServer
        /home/travis/gopath/src/github.com/brocaar/loraserver/internal/uplink/data/data.go:398 +0x815
```

brocaar, 2018-08-20:
* Which version are you using?
* Did you make any modifications?

brocaar, 2018-08-28:
I tried to reproduce your issue, but I'm unable to.... are you still experiencing this issue / do you have more hints how I can reproduce it?

sleepingAnt, 2018-08-28:
When I leave `"Network-servers -> TLS Certificates" in the lora-app-server web page` empty, everything is ok.
I think this is caused by that I didn't set TLS Certificates in `loraserver` config file, but used tls in the lora-app-server web page.

brocaar, 2018-08-28:
Hmm interesting. I would expect that you would get an error. Will test with that again. Please keep this issue open.

When you don't set any certificates in `loraserver.toml` and `lora-app-server.toml`, you should not enter any certificate information in LoRa App Server when adding the network-server.

sleepingAnt, 2018-08-28:
Okay, thanks man!

brocaar, 2018-08-29:
I'm able now to reproduce the issue. It happens when the CA certificate is blank, but the TLS cert and key are populated.

commit 36e9660b (2018-08-29): Fix panic on missing ca certificate.

Closes #349.
closed 2018-08-29 by brocaar