brocaar/chirpstack-network-server #352: New DownlinkTxPower Config Does Not Override Default Regional Downlink Power

alkindirizky, 2018-09-03:
### Is this a bug or a feature request?

(probably) bug

### What did you expect?

My Lora Gateway supports TX power up to 24 dBm. Setting TX power via adding this line to LoraServer config:

```
downlink_TX_power=24
```

is supposed to override the default TX power of my region, which is AS_923 (14 dBm)

### What happened?

Upon seeing the CONFIRMED downlink packet via LoraAppServer, under 'power' field inside 'txinfo' field, the power does not change to 24 dBm. It stays the same at default regional TX power, which 14 dBm.

However, the UNCONFIRMED MAC command downlink, **RXParamSetupReq**, shows correct TX power, 24 dBm. This issue also occurs at other TX power values that are supported by my gateway.

### What version are your using?

- loraserver : 2.1.0
- lora-app-server : 2.1.0
- lora-gateway-bridge : 2.5.1

### How can your issue be reproduced?

1. Change the downlink_tx_power config in loraserver.toml to any value supported by gateway, except -1
2. Start the LoraServer
3. Send a downlink packet

### Could you share your log output?

```
Sep 03 11:46:11  test-lora loraserver[2736]: time="2018-09-03T11:46:11Z" level=debug msg="running multicast scheduler batch"
Sep 03 11:46:11  test-lora loraserver[2736]: time="2018-09-03T11:46:11Z" level=debug msg="sql query executed" args="[100 2018-09-03 11:46:11.97948712 +0000 UTC m=+53.148443850]" duration="713.256µs" query="\n\t\tselect\n\t\t\t*\n\t\tfrom\n\t\t\tmulticast_queue\n\t\twhere\n\t\t\tschedule_at <= $2\n\t\torder by\n\t\t\tid\n\t\tlimit $1\n\t\tfor update skip locked\n\t"
Sep 03 11:46:12  test-lora loraserver[2736]: time="2018-09-03T11:46:12Z" level=debug msg="running class-b / class-c scheduler batch"
Sep 03 11:46:12  test-lora loraserver[2736]: time="2018-09-03T11:46:12Z" level=debug msg="sql query executed" args="[100 338891h46m32.122936747s]" duration=3.920491ms query="\n select\n     d.*\n from\n     device d\n inner join device_profile dp\n     on dp.device_profile_id = d.device_profile_id\n where (\n     \tdp.supports_class_c = true\n     \tor dp.supports_class_b = true\n     )\n     -- we want devices with queue items\n     and exists (\n  select\n      1\n  from\n      device_queue dq\n  where\n      dq.dev_eui = d.dev_eui\n      and (\n      \tdp.supports_class_c = true\n      \tor (\n      \t\tdp.supports_class_b = true\n      \t\tand dq.emit_at_time_since_gps_epoch <= $2\n      \t)\n      )\n     )\n     -- we don't want device with pending queue items that did not yet\n     -- timeout\n     and not exists (\n  select\n      1\n  from\n      device_queue dq\n  where\n      dq.dev_eui = d.dev_eui\n      and is_pending = true\n      and dq.timeout_after > now()\n     )\n order by\n     d.dev_eui\n limit $1\n for update of d skip locked"
Sep 03 11:46:12  test-lora loraserver[2736]: time="2018-09-03T11:46:12Z" level=error msg="schedule next device-queue item error" dev_eui=343437317436840e error="uplink gateway-history is empty"
Sep 03 11:46:12  test-lora loraserver[2736]: time="2018-09-03T11:46:12Z" level=debug msg="running multicast scheduler batch"
Sep 03 11:46:12  test-lora loraserver[2736]: time="2018-09-03T11:46:12Z" level=debug msg="sql query executed" args="[100 2018-09-03 11:46:12.981131845 +0000 UTC m=+54.150088594]" duration="668.561µs" query="\n\t\tselect\n\t\t\t*\n\t\tfrom\n\t\t\tmulticast_queue\n\t\twhere\n\t\t\tschedule_at <= $2\n\t\torder by\n\t\t\tid\n\t\tlimit $1\n\t\tfor update skip locked\n\t"
Sep 03 11:46:13  test-lora loraserver[2736]: time="2018-09-03T11:46:13Z" level=debug msg="running class-b / class-c scheduler batch"
Sep 03 11:46:13  test-lora loraserver[2736]: time="2018-09-03T11:46:13Z" level=debug msg="sql query executed" args="[100 338891h46m33.131319572s]" duration=6.016838ms query="\n select\n     d.*\n from\n     device d\n inner join device_profile dp\n     on dp.device_profile_id = d.device_profile_id\n where (\n     \tdp.supports_class_c = true\n     \tor dp.supports_class_b = true\n     )\n     -- we want devices with queue items\n     and exists (\n  select\n      1\n  from\n      device_queue dq\n  where\n      dq.dev_eui = d.dev_eui\n      and (\n      \tdp.supports_class_c = true\n      \tor (\n      \t\tdp.supports_class_b = true\n      \t\tand dq.emit_at_time_since_gps_epoch <= $2\n      \t)\n      )\n     )\n     -- we don't want device with pending queue items that did not yet\n     -- timeout\n     and not exists (\n  select\n      1\n  from\n      device_queue dq\n  where\n      dq.dev_eui = d.dev_eui\n      and is_pending = true\n      and dq.timeout_after > now()\n     )\n order by\n     d.dev_eui\n limit $1\n for update of d skip locked"
Sep 03 11:46:13  test-lora loraserver[2736]: time="2018-09-03T11:46:13Z" level=error msg="schedule next device-queue item error" dev_eui=343437317436840e error="uplink gateway-history is empty"
Sep 03 11:46:13  test-lora loraserver[2736]: time="2018-09-03T11:46:13Z" level=debug msg="running multicast scheduler batch"
Sep 03 11:46:13  test-lora loraserver[2736]: time="2018-09-03T11:46:13Z" level=debug msg="sql query executed" args="[100 2018-09-03 11:46:13.983144202 +0000 UTC m=+55.152100975]" duration="624.375µs" query="\n\t\tselect\n\t\t\t*\n\t\tfrom\n\t\t\tmulticast_queue\n\t\twhere\n\t\t\tschedule_at <= $2\n\t\torder by\n\t\t\tid\n\t\tlimit $1\n\t\tfor update skip locked\n\t"
Sep 03 11:46:14  test-lora loraserver[2736]: time="2018-09-03T11:46:14Z" level=debug msg="running class-b / class-c scheduler batch"
Sep 03 11:46:14  test-lora loraserver[2736]: time="2018-09-03T11:46:14Z" level=debug msg="sql query executed" args="[100 338891h46m34.147930005s]" duration=4.611676ms query="\n select\n     d.*\n from\n     device d\n inner join device_profile dp\n     on dp.device_profile_id = d.device_profile_id\n where (\n     \tdp.supports_class_c = true\n     \tor dp.supports_class_b = true\n     )\n     -- we want devices with queue items\n     and exists (\n  select\n      1\n  from\n      device_queue dq\n  where\n      dq.dev_eui = d.dev_eui\n      and (\n      \tdp.supports_class_c = true\n      \tor (\n      \t\tdp.supports_class_b = true\n      \t\tand dq.emit_at_time_since_gps_epoch <= $2\n      \t)\n      )\n     )\n     -- we don't want device with pending queue items that did not yet\n     -- timeout\n     and not exists (\n  select\n      1\n  from\n      device_queue dq\n  where\n      dq.dev_eui = d.dev_eui\n      and is_pending = true\n      and dq.timeout_after > now()\n     )\n order by\n     d.dev_eui\n limit $1\n for update of d skip locked"
Sep 03 11:46:14  test-lora loraserver[2736]: time="2018-09-03T11:46:14Z" level=error msg="schedule next device-queue item error" dev_eui=343437317436840e error="uplink gateway-history is empty"
Sep 03 11:46:14  test-lora loraserver[2736]: time="2018-09-03T11:46:14Z" level=debug msg="sql query executed" args="[3930323560378704]" duration="397.448µs" query="\n select\n     *\n from\n     device_queue\n where\n     dev_eui = $1\n order by\n     f_cnt"
Sep 03 11:46:14  test-lora loraserver[2736]: time="2018-09-03T11:46:14Z" level=info msg="finished unary call with code OK" grpc.code=OK grpc.method=GetNextDownlinkFCntForDevEUI grpc.service=ns.NetworkServerService grpc.start_time="2018-09-03T11:46:14Z" grpc.time_ms=0.723 peer.address="127.0.0.1:49140" span.kind=server system=grpc
Sep 03 11:46:14  test-lora loraserver[2736]: time="2018-09-03T11:46:14Z" level=debug msg="sql query executed" args="[[57 48 50 53 96 55 135 4]]" duration="412.017µs" query="select * from device where dev_eui = $1"
Sep 03 11:46:14  test-lora loraserver[2736]: time="2018-09-03T11:46:14Z" level=debug msg="sql query executed" args="[9892786f-6253-4a04-acf7-1704ef33673a]" duration="714.807µs" query="\n select\n     created_at,\n     updated_at,\n\n     device_profile_id,\n     supports_class_b,\n     class_b_timeout,\n     ping_slot_period,\n     ping_slot_dr,\n     ping_slot_freq,\n     supports_class_c,\n     class_c_timeout,\n     mac_version,\n     reg_params_revision,\n     rx_delay_1,\n     rx_dr_offset_1,\n     rx_data_rate_2,\n     rx_freq_2,\n     factory_preset_freqs,\n     max_eirp,\n     max_duty_cycle,\n     supports_join,\n     rf_region,\n     supports_32bit_fcnt\n from device_profile\n where\n     device_profile_id = $1\n "
Sep 03 11:46:14  test-lora loraserver[2736]: time="2018-09-03T11:46:14Z" level=debug msg="sql query executed" args="[2018-09-03 11:46:14.626226077 +0000 UTC m=+55.795182798 2018-09-03 11:46:14.626226077 +0000 UTC m=+55.795182798 [57 48 50 53  [... cut ...] ng queue items that did not yet\n     -- timeout\n     and not exists (\n  select\n      1\n  from\n      device_queue dq\n  where\n      dq.dev_eui = d.dev_eui\n      and is_pending = true\n      and dq.timeout_after > now()\n     )\n order by\n     d.dev_eui\n limit $1\n for update of d skip locked"
Sep 03 11:46:18  test-lora loraserver[2736]: time="2018-09-03T11:46:18Z" level=error msg="schedule next device-queue item error" dev_eui=343437317436840e error="uplink gateway-history is empty"
```

andrewflash, 2018-09-06:
Any update regarding this issue?

brocaar, 2018-09-06:
I think I have found the issue. It wasn't correctly overwriting the TX power value for Class B and C.

commit 6f053528 (2018-09-06): Fix tx-power override for Class B and C.

Closes #352.
closed 2018-09-06 by brocaar