brocaar/chirpstack-network-server #353: high frequency queuing

emmanuel-florent, 2018-09-15:
### Is this a bug or a feature request?
This is a feature request
### What did you expect?
For my project it's mandatory to send two *very* consecutive packets. I am somehow limited by the queue pooling frequency of one second.

### What happened?
This parameters are hardcoded in `/internal/config/config.go` 
```
// SchedulerInterval it the interval in which the a scheduler
// must run.
var SchedulerInterval = time.Second

// ClassCDownlinkLockDuration contains the duration to lock the downlink
// Class-C transmissions after a preceeding downlink tx.
var ClassCDownlinkLockDuration = time.Second * 2
```
Mayn I submit a PR that make parametrisable or not.
Does it make to sense to you guys or not ?

emmanuel-florent, 2018-09-17:
The project bundle two packet for a single ModBus command and get one packet answer.
It have to works with 30 nodes per channels every minutes. By diminishing the `SchedulerInterval` to `50ms` and adding a queue sequencer I can archive round trip times of `80ms` instead of `2000ms`
In my project schedule a FiFo queue using time on air calculation to prevent part of packet loss.

brocaar, 2018-09-20:
Have you already tried if it is possible to emit the LoRaWAN frame in 50ms? See also https://www.thethingsnetwork.org/forum/t/spreadsheet-for-lora-airtime-calculation/1190.

Apart from that, please do change this into a config variable. I think it would be good to add a `Scheduler` struct under: https://github.com/brocaar/loraserver/blob/master/internal/config/config.go#L55 for the generic (not Class B / C specific) values. Then for class specific values, add these under `ClassB` (https://github.com/brocaar/loraserver/blob/master/internal/config/config.go#L74) or create a new `ClassC` struct if needed.

Under https://github.com/brocaar/loraserver/blob/master/cmd/loraserver/cmd/root.go#L67 you can define the default values (I would suggest to keep the current hardcoded values as defaults).

closed 2018-10-02 by brocaar

sp193, 2020-06-11:
Pardon me for bringing up this closed issue, but I must ask - does the Semtech packet forwarder not help with enqueuing class C downlink packets, in a way that avoids collision?

The Basic Station LNS protocol appears to have this behaviour built into its design, while the classic Semtech UDP Packet Forwarder appears to have downlink scheduling implemented later on.

Starting with the [Semtech UDP Packet Forwarder release v3.0.0](https://github.com/Lora-net/packet_forwarder/releases/tag/v3.0.0) on 06/06/2016, a "Just-in-time" downlink scheduler was added. It appears to automatically find an available slot for the new frame, if the "imme" bit is set. The scheduling is done within this block: https://github.com/Lora-net/packet_forwarder/blob/master/lora_pkt_fwd/src/jitqueue.c#L173

I regret to say that I cannot commit to test to see if it does work as I expect. So I thought I should ask if you have experienced something that has proved otherwise.