brocaar/chirpstack-network-server #358: loraserver not starting, appears to complain about an SQL error

jcwren, 2018-10-28:
I've been running a >really< old version of loraserver and friends, and decided to upgrade it this weekend. I see 2.6.0 is the most recent version, but 2.3.0 is the most recent that apt-get will pull. Postgresql is version 9.4.

### Is this a bug or a feature request?
Bug

### What did you expect?
loraserver to start

### What happened?
loraserver didn't start (errors below)

### What version are your using?
2.3.0 (most recent for apt-get with raspian)

### How can your issue be reproduced?
Fresh raspian SD card, install loraserver, configure and start it

### Could you share your log output?
Oct 28 22:24:52 rpi3 systemd[1]: Started LoRa Server.
Oct 28 22:24:52 rpi3 loraserver[8478]: time="2018-10-28T22:24:52Z" level=info msg="starting LoRa Server" band=US_902_928 docs="https://docs.loraserver.io/" net_id=000000 version=2.3.0
Oct 28 22:24:52 rpi3 loraserver[8478]: time="2018-10-28T22:24:52Z" level=info msg="disabling all channels"
Oct 28 22:24:52 rpi3 loraserver[8478]: time="2018-10-28T22:24:52Z" level=info msg="enabling channels" channels="[0 1 2 3 4 5 6 7]"
Oct 28 22:24:52 rpi3 loraserver[8478]: time="2018-10-28T22:24:52Z" level=info msg="setup redis connection pool" url="redis://localhost:6379"
Oct 28 22:24:52 rpi3 loraserver[8478]: time="2018-10-28T22:24:52Z" level=info msg="connecting to postgresql"
Oct 28 22:24:52 rpi3 loraserver[8478]: time="2018-10-28T22:24:52Z" level=info msg="gateway/mqtt: connecting to mqtt broker" server="tcp://localhost:1883"
Oct 28 22:24:52 rpi3 loraserver[8478]: time="2018-10-28T22:24:52Z" level=info msg="no geolocation-server configured"
Oct 28 22:24:52 rpi3 loraserver[8478]: time="2018-10-28T22:24:52Z" level=info msg="configuring join-server client" ca_cert= server="http://localhost:8003" tls_cert= tls_key=
Oct 28 22:24:52 rpi3 loraserver[8478]: time="2018-10-28T22:24:52Z" level=info msg="no network-controller configured"
Oct 28 22:24:52 rpi3 loraserver[8478]: time="2018-10-28T22:24:52Z" level=info msg="applying database migrations"
Oct 28 22:24:52 rpi3 loraserver[8478]: time="2018-10-28T22:24:52Z" level=info msg="backend/gateway: connected to mqtt server"
Oct 28 22:24:52 rpi3 loraserver[8478]: time="2018-10-28T22:24:52Z" level=info msg="gateway/mqtt: subscribing to rx topic" qos=0 topic=gateway/+/rx
Oct 28 22:24:52 rpi3 loraserver[8478]: time="2018-10-28T22:24:52Z" level=info msg="gateway/mqtt: subscribing to stats topic" qos=0 topic=gateway/+/stats
Oct 28 22:24:52 rpi3 loraserver[8478]: time="2018-10-28T22:24:52Z" level=info msg="backend/gateway: subscribing to ack topic" qos=0 topic=gateway/+/ack
Oct 28 22:24:52 rpi3 loraserver[8478]: time="2018-10-28T22:24:52Z" level=info msg="migrations applied" count=0
Oct 28 22:24:52 rpi3 loraserver[8478]: time="2018-10-28T22:24:52Z" level=debug msg="sql query executed" args="[]" duration="540.365µs" query="lock table code_migration"
Oct 28 22:24:52 rpi3 loraserver[8478]: time="2018-10-28T22:24:52Z" level=debug msg="sql query executed" args="[v1_to_v2_flush_profiles_cache 2018-10-28 22:24:52.494088763 +0000 UTC m=+0.058419340]" duration="586.979µs" query="\n\t\t\tinsert into code_migration (\n\t\t\t\tid,\n\t\t\t\tapplied_at\n\t\t\t) values ($1, $2)\n\t\t\ton conflict\n\t\t\t\tdo nothing\n\t\t"
Oct 28 22:24:52 rpi3 loraserver[8478]: time="2018-10-28T22:24:52Z" level=fatal msg="pq: syntax error at or near \"on\""
Oct 28 22:24:52 rpi3 systemd[1]: loraserver.service: main process exited, code=exited, status=1/FAILURE
Oct 28 22:24:52 rpi3 systemd[1]: Unit loraserver.service entered failed state.
Oct 28 22:24:52 rpi3 systemd[1]: loraserver.service holdoff time over, scheduling restart.
Oct 28 22:24:52 rpi3 systemd[1]: Stopping LoRa Server...
Oct 28 22:24:52 rpi3 systemd[1]: Starting LoRa Server...
Oct 28 22:24:52 rpi3 systemd[1]: loraserver.service start request repeated too quickly, refusing to start.
Oct 28 22:24:52 rpi3 systemd[1]: Failed to start LoRa Server.
Oct 28 22:24:52 rpi3 systemd[1]: Unit loraserver.service entered failed state.

jcwren, 2018-10-28:
Looking through the closed comments, I found #171. I knew I didn't like postgresql all these years for a good reason...

brocaar, 2018-10-29:
Please refer to https://www.loraserver.io/loraserver/install/requirements/. PostgreSQL 9.5+ is required.

closed 2018-10-29 by brocaar

jcwren, 2018-10-29:
Alas, https://www.loraserver.io/guides/debian-ubuntu/ says nothing about the required version, which was where I started.

brocaar, 2018-10-30:
But it does mention the Ubuntu / Debian distribution versions it has been tested on. The tested versions all contain PostgreSQL 9.5+ in the repository :)

luckshot, 2018-11-15:
Following the guide word-for-word from https://www.loraserver.io/guides/debian-ubuntu/ does not install the correct version of PostgreSQL when using Debian Jessie.  Debian Jessie repository provides PostgreSQL 9.4.18.

luckshot, 2018-11-15:
FWIW Debian Jessie (Debian 8), provides PostgreSQL 9.4.18, which will not work.  Debian Stretch is required.

brocaar, 2018-11-15:
Stretch is not required but you need to perform extra steps to install
PostgreSQL 9.5+