brocaar/chirpstack-network-server #362: SpreadingFactor metadata seem to be missing from metadata

fiskfan, 2018-12-04:
I have an issue where the SpreadingFactor information seem to be missing from the metadata. When googling, i find information that it normally exists.

I get some of the metadata from my gateway and it looks like this:
[image]

As seen, the SF is not visible. What might be the reason for that?

I have some pervious experience from TTN, and the TTN metadata also airtime like this:
        "data_rate": "SF7BW125",
        "airtime": 66816000,
        "coding_rate": "4/5",

The SF information would be helpful determining the LoRa network coverage and the airtime could also be nice to get help with listing devices that has poor coverage or is misbehaving once we scale up in number of nodes.

fiskfan, 2018-12-04:
Okey. So after digging more in this i find that the "Live Lorawan frames" tab shows the spreading factor. But this information is not seen under "Live device data" nor in the MQTT packet.

acourt, 2018-12-04:
The spreading factor is a function of the datarate.

fiskfan, 2018-12-04:
Thanks for your quick reply.

The datarate and/or SF does not exist as seen in the screenshot i attached. I would been nice to have them in the metadata for each message that the device send and not only in the gw metadata.

When doing integrations im getting the device messages on mqtt and want to have the SF/datarate info for each message so that i can add this to the metadata in the portal im using for visualisation

acourt, 2018-12-04:
The datarate is under **_txInfo_** as "_**dr**_". In your screen shot the DR is 5.

With respect to the RSSI information, it gets added to the MQTT packet if you toggle the checkmark "Add gateway meta-data" under Service Profile when you create your service.

closed 2018-12-09 by brocaar