brocaar/chirpstack-network-server #370: Monitor Duty Cycle On Gateways

tomaql, 2019-02-11:
### Is this a bug or a feature request?
Feature Request

### What did you expect?
I need a way to monitor the duty cycle on a given gateway.

We currently monitor them using the last seen at to ensure they are still alive but need something more robust.

### What happened?
Nothing

### What version are your using?
Latest

### How can your issue be reproduced?

### Could you share your log output?

brocaar, 2019-02-13:
Currently this information is not exposed by the packet-forwarder. Some work is being done on new packet-forwarders (e.g. the new Semtech packet-forwarder) which could facilitate this. In my opinion, the duty-cycle needs to be calculated by the gateway, not the network-server.

queifaro, 2020-05-28:
Hello,
are some implementation (over UDP packet forwarder) already done on this feature ? or is the feature in plan for the future ?

Thanks 
Best
Fabio

brocaar, 2020-10-05:
I have created the above issue ^^

closed 2020-10-05 by brocaar