brocaar/chirpstack-network-server #371: Class C MAC command DeviceModeInd not handled

skywalker999u, 2019-02-12:
### Is this a bug or a feature request?
bug

### What did you expect?
loraserver handling LoRaWAN v1.1 MAC command DeviceModeInd and payload (0x20 0x02) from node to indicate its mode changing to class C

### What happened?
lora-app-server shows that loraserver failed to recognize DeviceModeInd (shows "CID(32)") and recognized payload (0x02) as MAC command LinkCheckReq instead
Also, from loraserver/internal/maccommand/maccommand.go, there is no case for handling DeviceModeInd.

### What version are your using?
2.4.1 (with lora-app-server 2.4.1)

### How can your issue be reproduced?
add a LoRaWAN v1.1 compliant class C node and make it send a DeviceModeInd MAC command

### Could you share your log output?
From raw lorawan frame:
```
[
    {
        "downlinkMetaData": {
            "txInfo": {
                "gatewayId": [gwid],
                "immediately": false,
                "timeSinceGpsEpoch": null,
                "timestamp": 2893638052,
                "frequency": 923200000,
                "power": 14,
                "modulation": "LORA",
                "loraModulationInfo": {
                    "bandwidth": 125,
                    "spreadingFactor": 8,
                    "codeRate": "4/5",
                    "polarizationInversion": true
                },
                "board": 0,
                "antenna": 0
            }
        },
        "phyPayload": {
            "mhdr": {
                "mType": "UnconfirmedDataDown",
                "major": "LoRaWANR1"
            },
            "macPayload": {
                "fhdr": {
                    "devAddr": [devaddr],
                    "fCtrl": {
                        "adr": true,
                        "adrAckReq": false,
                        "ack": false,
                        "fPending": false,
                        "classB": false
                    },
                    "fCnt": 4,
                    "fOpts": [
                        {
                            "cid": "LinkCheckReq",
                            "payload": {
                                "margin": 20,
                                "gwCnt": 1
                            }
                        }
                    ]
                },
                "fPort": null,
                "frmPayload": null
            },
            "mic": [mic]
        }
    },
    {
        "uplinkMetaData": {
            "rxInfo": [
                {
                    "gatewayId": [gwid],
                    "time": "1970-01-01T22:16:47.217Z",
                    "timeSinceGpsEpoch": null,
                    "timestamp": 2892638052,
                    "rssi": -30,
                    "loraSnr": 10.8,
                    "channel": 0,
                    "rfChain": 0,
                    "board": 0,
                    "antenna": 0,
                    "location": {
                        "latitude": 0,
                        "longitude": 0,
                        "altitude": 0,
                        "source": "UNKNOWN",
                        "accuracy": 0
                    },
                    "fineTimestampType": "NONE"
                }
            ],
            "txInfo": {
                "frequency": 923200000,
                "modulation": "LORA",
                "loRaModulationInfo": {
                    "bandwidth": 125,
                    "spreadingFactor": 8,
                    "codeRate": "4/5",
                    "polarizationInversion": false
                }
            }
        },
        "phyPayload": {
            "mhdr": {
                "mType": "UnconfirmedDataUp",
                "major": "LoRaWANR1"
            },
            "macPayload": {
                "fhdr": {
                    "devAddr": [devaddr],
                    "fCtrl": {
                        "adr": false,
                        "adrAckReq": false,
                        "ack": false,
                        "fPending": false,
                        "classB": false
                    },
                    "fCnt": 2,
                    "fOpts": [
                        {
                            "cid": "CID(32)",
                            "payload": null
                        },
                        {
                            "cid": "LinkCheckReq",
                            "payload": null
                        }
                    ]
                },
                "fPort": null,
                "frmPayload": null
            },
            "mic": [mic]
        }
    }
]
```

brocaar, 2019-02-13:
You're right, this mac-command is not yet handled. I'll fix this asap :-)

closed 2019-02-18 by brocaar

skywalker999u, 2019-02-19:
Thanks for the speedy fix.
However there is still a bug.
According to v1.1 spec, DeviceModeConf should have the payload same as that came from DeviceModeInd.

brocaar, 2019-02-19:
You're right. This was also missing the `brocaar/lorawan` package. I have just implemented this payload and will also fix this in loraserver: https://github.com/brocaar/lorawan/commit/e320612b3e3d0c78fa1bf88c26e270617f0a794b.

skywalker999u, 2019-02-20:
Thanks for the implementation/fix.

I am not sure what I get now is an issue about loraserver or the device,
but after I ran the compiled "./build/loraserver" and tried DeviceModeInd-ing the device as class C, 

I saw from the raw frame that the payload is still handled separately from DeviceModeInd, and without DeviceModeConf downlink response. (lora-app-server raw frame posted at bottom, and not yet updated so it is still showing CID(32) :D )
Tho I did once saw lora-app-server raw frame saying loraserver has replied a DeviceModeConf without payload,
and I saw the following error output from the program:
> ERRO[0128] read pending mac-command error: gob decode error: not enough remaining bytes  cid=DeviceModeInd dev_eui=[deveui]

which the 2 results indicate that DeviceModeInd is being handled, but might not be handled correctly....?

Raw Frame:
```
[
    {
        "uplinkMetaData": {
            "rxInfo": [
                {
                    "gatewayId": [deveui],
                    "time": "1970-01-02T02:56:16.027Z",
                    "timeSinceGpsEpoch": null,
                    "timestamp": 2481593236,
                    "rssi": -36,
                    "loraSnr": 11.3,
                    "channel": 2,
                    "rfChain": 0,
                    "board": 0,
                    "antenna": 0,
                    "location": {
                        "latitude": 0,
                        "longitude": 0,
                        "altitude": 0,
                        "source": "UNKNOWN",
                        "accuracy": 0
                    },
                    "fineTimestampType": "NONE"
                }
            ],
            "txInfo": {
                "frequency": 923600000,
                "modulation": "LORA",
                "loRaModulationInfo": {
                    "bandwidth": 125,
                    "spreadingFactor": 8,
                    "codeRate": "4/5",
                    "polarizationInversion": false
                }
            }
        },
        "phyPayload": {
            "mhdr": {
                "mType": "UnconfirmedDataUp",
                "major": "LoRaWANR1"
            },
            "macPayload": {
                "fhdr": {
                    "devAddr": [devaddr],
                    "fCtrl": {
                        "adr": false,
                        "adrAckReq": false,
                        "ack": false,
                        "fPending": false,
                        "classB": false
                    },
                    "fCnt": 2,
                    "fOpts": [
                        {
                            "cid": "CID(32)",
                            "payload": null
                        },
                        {
                            "cid": "LinkCheckReq",
                            "payload": null
                        }
                    ]
                },
                "fPort": null,
                "frmPayload": null
            },
            "mic": "b532de6b"
        }
    }
]
```

commit ca2af994 (2019-02-18): Refactor device-mode (A/B/C) tracking.

LoRaWAN 1.1 provides a mac-command to let the device signal
to the network-server that it has changed between Class A & C.
Note that the change to Class-B was already possible for LoRaWAN 1.0
devices by setting the ClassB flag.

Closes #371.