brocaar/chirpstack-network-server #376: Multiple Bands

temporaryaccount, 2019-02-26:
### Is this a bug or a feature request?

Feature request

### What did you expect?

We are testing both EU_433 and EU_863_870 band gateways, but the configuration file wants to be provided with only *one* band name. It would be nice if we could provide the band per gateway/gateway profile in the GUI instead of providing it globally in the configuration file. Thank you.

### What version are your using?

2.5.0

brocaar, 2019-05-14:
The current suggested approach for this is to setup multiple LoRa Server instances, one for each region. This also allows you to tune the `loraserver.toml` configuration specifically for that region (e.g. RX delay, DR offset, ....).

Note that you can start multiple LoRa Server instances simultaneously and that LoRa App Server can connect to multiple LoRa Server instances so that you can manage multiple regions from a single web-interface.

closed 2019-06-24 by brocaar

temporaryaccount, 2019-06-28:
Thank you, sorry for the late response.