brocaar/chirpstack-network-server #378: ADR DR backoff

kuripiiritus, 2019-02-27:
### Is this a bug or a feature request?
Bug
### What did you expect?
System to command node to set a lower DR value (higher SF)
### What happened?
Only commands to set TX power to lower value (higher dBm)
### What version are your using?
LoRa Server: v2.3.0
LoRa App Server: v2.4.0
### How can your issue be reproduced?
Let ADR do it's magic and then create bad reception for the node
### Could you share your log output?

Currently we have been looking trough the ADR code on server side and it seems that it only has code for increasing the TXpower but not the SF when the TXpower has been maxed out and  SNR is still under the threshold

brocaar, 2019-02-27:
I'm not sure if it is a good practice to let the NS lower the data-rate as this also increase the airtime of the device(s). When the reception is bad because of collisions, this means lowering the DR will make it even worse triggering the NS to lower the DR once again making it worse again, ... That is at least what I have learned from others.

Please note that a mechanism for lowering the DR is already implemented at the device-side as specified by the LoRaWAN specification.

kuripiiritus, 2019-02-27:
Yes, there is a mechanism in place on the device side. Network side adjust would be fixing it before it actually happens, as usually network will know before the node, that things are getting out of hand (confirmed messages vs all messages). 

But i can understand the reasoning behind deliberately not touching it on NS side. Now we know that we understand the code correctly and it is as it seems.

jamesl-dm, 2019-02-28:
Wow - I didn't realize network servers were set up that way. Seems problematic.

For a device sending unconfirmed messages with the default ADR timing parameters, you'd need to lose 32 messages in a row before the DR backoff happened. And the latency could be as bad as 96 messages. Imagine if the link margin degraded to the point that 90% of packets were being lost. You'd still probably get 3 successful transmissions before your run of 32 failures completed. So the node would be stuck with 90% packet loss for quite some time.

Imagine the node is transmitting only 12 messages a day. If the link becomes marginal, then dies, it could take 8 days before the DR backoff happens. The ADR_ACK_DELAY scheme with the default numbers has always seemed a bit of a problematic mechanism to me. But if the server doesn't preemptively fix the link margin, it's seems doubly impractical.

I'm not saying leaving the backoff to the node is wrong - considering the limited uplink capacity at SF12, it seems pretty reasonable. But maybe it would be better to differentiate between a strong signal with a high noise floor, and a weak signal that will need a lower SF just to compete with the thermal noise? If the signal is genuinely too weak, I think the gateway should lower the datarate.

One approach would be to calculate the estimated signal power (same as RSSI if SNR >0, RSSI + SNR if SNR < 0), and compare it to the datasheet sensitivity:
[image]

If the estimated power is lower than the sensitivity + required margin, then it's time to reduce the DR. In the event that interference is drowning out the packets (so estimated power is above threshold), or the packets simply aren't arriving at all, you could leave the DR reduction to the device.

brocaar, 2019-03-01:
Please note that ADR should be used for "static" devices. Using the `installation_margin` setting it is possible to configure a margin that the network keeps in mind when increasing the DR or lowering the TX Power. Increasing this `installation_margin` will make the network more pessimistic, but you will be more at the safe side.

Also note that a device is free to validates its connectivity with the network using the `LinkCheckReq` mac-command.

closed 2019-06-26 by brocaar