brocaar/chirpstack-network-server #382: LoRa redis server error

sarju32, 2019-03-11:
Hi,

Attached is issue faced in LoRa.

Issue is not observed when everything in the Database is removed and added again.
After this when board reboots without deleting the database this issue is obtained.

Thanks & Regards,
Sarjoon

brocaar, 2019-03-18:
Attachment is missing.

sarju32, 2019-03-18:
Hi Orne Brocaar,

We tried to attach the image in Bug or Feature request page. Its not 
working,

So we attach the image along with this mail.

Thanks,

sarju32, 2019-03-18:
[image]

brocaar, 2019-03-18:
This is not a bug. It looks like your (Redis) database is not started on a re-boot and thus LoRa Server complains that it can't connect.

closed 2019-03-18 by brocaar