brocaar/chirpstack-network-server #383: Yet another Duty cycle question

Timo69, 2019-03-12:
### Is this a bug or a feature request?

Feature request.

### What did you expect?

We would want a way to enforce gateways downlink duty cycle.

### What happened?

Nothing

### What version are your using?

Latest
### How can your issue be reproduced?

### Could you share your log output?

We would want a way to enforce gateway downlink duty cycle. We've seen on the issue #370 and #70 that there's some discussion regarding which components should have the responsability of handling duty cycle. It seems that there is a consensus about having the packet forwarder computing the duty cycle itself and reporting it to the loraserver. Is it open for discussion or should we wait for future packet forwarder implementation/version ?

We thought about implementing an estimation of this logic on the loraserver side to load balance frames between gateways (and not only send packets to the best gateway sorted by snr and rssi) ?

brocaar, 2019-04-08:
I'm working on a refactor of LoRa Gateway Bridge, to also add support for the Basic Station packet-forwarder. I would prefer to move the discussion to this repository and see if we can implement it there in the future :-)

Maybe LoRa Server would need some additional logic to depend which gateway to use for the downlink based on duty-cycle limitations, but in my opinion the gateway (or LoRa Gateway Bridge?) is responsible for calculating the duty-cycle.

urbie-mk2, 2019-06-10:
I can agree that the blancing part of commanding gateway transmissions belongs more to the loraserver than to the gateway:

If a gateway is only used by one loraserver (more are possible) the info about load can be calculated in the loraserver alone.

If a gateway is used by more than one loraserver / network server (lora-packet-multiplexer) the gateway must produce info about its load / usage and communicate this information to the loraserver(s).

In a private (small) infrastructure the gateway <-> network server relation is known but this would be a special case and thus I think it should be the gateways task to report more information to the loraserver / network server for this purpose.

Loraserver can then use the information about load (and other variables like region, duty cycle limit, used frequency bands, usage of listen before talk, half or full duplex, number of available channels, number of antennas) as input to a function to choose a gateway for transmission.

I think the gateway bridge should not be used as target for this implementation. The Loraserver already decides about many parameters to be sent for transmission by the gateways (dr, inverted iq, frequency, transmit power, ...), the task of the gateway-bridge should remain accumulating / transforming from udp to mqtt.

brocaar, 2020-10-05:
I have created the above issue ^^

closed 2020-10-05 by brocaar