brocaar/chirpstack-network-server #384: At OTAA, LoRa Server returns Unconffrimed data down to Confirmed data.

sarju32, 2019-03-18:
#Bug
When LoRa node sends unconfirmed data no reply is sent from server immediately, but after sending confirmed data from LoRa node LoRa server sends unconfirmed data.

brocaar, 2019-03-18:
Which looks valid to me as LoRa Server needs to acknowledge the confirmed uplink.

closed 2019-03-18 by brocaar

sarju32, 2019-03-18:
Please refer below attached LoRa_sequence.png ..
[image]

The sequence diagram is given, For ConfirmedUp we are not receiving ConfirmedDown Data rather we are seeing UnconfirmedDown. 
Any reason for the attached Is there some delay 

Attached the images that was taken on LoRaWAN packet decoder for each sequence.
[image]
[image]
[image]
[image]
[image]
[image]