brocaar/chirpstack-network-server #385: Gateway discover problem after update from loraserver 2.5.0 to 2.6.0 and lora-app-server to version 2.5.1

dlamboglia, 2019-03-21:
I have 3 gateways with the 2.5.0 version of the LoraServer all 3 gateways were seen with Gateway discovery, updating to the 2.6.0 and 2.5.1 lora-app-server version are seen only in pairs.

dlamboglia, 2019-03-22:
in debug loraserver view a error:

time="2019-03-22T10:09:17+01:00" level=info msg="gateway/mqtt: publishing downlink frame" qos=2 topic=gateway/001bc506702009a9/tx
time="2019-03-22T10:09:17+01:00" level=info msg="finished unary call with code OK" grpc.code=OK grpc.method=SendProprietaryPayload grpc.service=ns.NetworkServerService grpc.start_time="2019-03-22T10:09:17+01:00" grpc.time_ms=0.898 peer.address="51.75.68.135:55332" span.kind=server system=grpc
time="2019-03-22T10:09:19+01:00" level=info msg="gateway/mqtt: uplink frame received"
time="2019-03-22T10:09:20+01:00" level=info msg="finished client unary call" grpc.code=OK grpc.method=HandleProprietaryUplink grpc.service=as.ApplicationServerService grpc.time_ms=9.325 span.kind=client system=grpc
time="2019-03-22T10:09:20+01:00" level=info msg="backend/gateway: downlink tx acknowledgement received" gateway_id=001bc506702009a9
time="2019-03-22T10:09:20+01:00" level=info msg="gateway/mqtt: uplink frame received"
time="2019-03-22T10:09:20+01:00" level=info msg="gateway/mqtt: uplink frame received"
time="2019-03-22T10:09:20+01:00" level=error msg="finished client unary call" error="rpc error: code = Internal desc = handle received ping error: get ping lookup error: get ping lookup error: redigo: nil returned" grpc.code=Internal grpc.method=HandleProprietaryUplink grpc.service=as.ApplicationServerService grpc.time_ms=0.981 span.kind=client system=grpc
time="2019-03-22T10:09:20+01:00" level=error msg="handle proprietary up error" error="rpc error: code = Internal desc = handle received ping error: get ping lookup error: get ping lookup error: redigo: nil returned"
time="2019-03-22T10:09:21+01:00" level=info msg="gateway/mqtt: gateway stats packet received" gateway_id=001bc506702009a9

brocaar, 2019-04-08:
I expect this has been fixed by v2.6.1 :-)

closed 2019-04-08 by brocaar