brocaar/chirpstack-network-server #388: Answer to Sticky MAC commands like RXTimingSetupReq are not ack by loraserver

Regimbal, 2019-04-04:
### Is this a bug or a feature request?
I'd say a bug or at least improvement required

### What did you expect?
From reading the spec (p.40 of 1.1, 30 of 1.0.2)
> The RXTimingSetupAns command should be added in the FOpt field of all uplinks until a
class A downlink is received by the end-device.

From what I understand, the server shall acknowledge the device once it gets the RXTimingSetupAns just by sending it a downlink. 

### What happened?
If my device sends only unconfirmed messages, then it will keep inserting the RXTimingSetupAns in its further uplink messages.

### What version are your using?
2.6.0

### How can your issue be reproduced?
Just configure device to send unconfirmed messages

### Could you share your log output?

brocaar, 2019-04-17:
Thanks, I understand what you mean and I think you're right.

Let me think what would be the best way to implement this. On the downlink context there is already a `MustSend` bool which triggers a downlink, even when there are no mac-commands / FRMPayload to send. Currently this can not be set from the uplink context.

What would be needed I believe

- [ ] handle uplink mac-command function returns if a downlink must be send
- [ ]  `datadown.HandleResponse` must contain a `mustSend` argument

brocaar, 2019-05-15:
This has been fixed by the above commit. Thanks again for reporting :+1:

mluis1, 2019-05-15:
There are other MAC commands similar to RxTimmingSetupReq. Looking at the changes you provide it looks like they are not handled neither.

The other MAC commands are:
```c
static bool IsSticky( uint8_t cid )
{
    switch( cid )
    {
        case MOTE_MAC_RESET_IND:
        case MOTE_MAC_REKEY_IND:
        case MOTE_MAC_DEVICE_MODE_IND:
        case MOTE_MAC_DL_CHANNEL_ANS:
        case MOTE_MAC_RX_PARAM_SETUP_ANS:
        case MOTE_MAC_RX_TIMING_SETUP_ANS:
            return true;
        default:
            return false;
    }
}
```
There will be one more to be added as soon as the LoRa-Alliance issues an ERRATA note concerning `TxParamSetupAns` MAC command which should also behave in the same way.