brocaar/chirpstack-network-server #389: adrAckReq response from server has empty fOpts

rayman18, 2019-04-10:
### Is this a bug or a feature request?
Bug

### What did you expect?
The adrAckReq response from server should have LinkADRReq info inside fOpts with dataRate and txPower info.

### What happened?
The server answers but dowlink has field fOpts:null
No ADR info in downlink.

### What version are your using?
2.6.1

### How can your issue be reproduced?
After the node counter AdrAckCounter reach EU868_ADR_ACK_LIMIT(64) the node sends adrAckReq:true
Then the server sends downlink but without ADR info

### Could you share your log output?
Attached uplink and downlink frames log

[gateway-undefined.txt](https://github.com/brocaar/loraserver/files/3063649/gateway-undefined.txt)

brocaar, 2019-04-11:
The `ADRACKReq` must trigger a downlink, even when there is no payload or mac-commands to send to confirm the network is still able to receive the device-data. I don't believe it must send a `LinkADRReq` mac-command.

[image]

The network is required to respond with a downlink frame .... any received downlink frame following an uplink frame resets the `ADR_ACK_CNT`.

closed 2019-04-11 by brocaar

rayman18, 2019-04-11:
@brocaar 
> The network is required to respond with a downlink frame .... any received downlink frame following an uplink frame resets the `ADR_ACK_CNT`.

The node reset the `ADR_ACK_CNT`, but the problem is the following code in the node:

[image]

When the `ADR_ACK_CNT `reaches the `EU868_ADR_ACK_LIMIT`(64), the txPower is set to `EU868_MAX_TX_POWER`(0).

Then `ADRACKReq` is sent to server and server respond with a downlink frame, but without LinkADRReq mac-command. 
This way the node reamains using `EU868_MAX_TX_POWER`(0) forever. 

So the objective of ADR scheme is being enabled to increase the battery life of the end-device and maximize the network capacity. But in this case, the battery life is not optimized.

brocaar, 2019-04-11:
> When the ADR_ACK_CNT reaches the EU868_ADR_ACK_LIMIT(64), **the txPower is set to EU868_MAX_TX_POWER(0)**.

Why is the device doing this? I'm not aware that this is in line with the LoRaWAN specifications. If you believe it is, could you refer to the section describing this?

rayman18, 2019-04-11:
> 
> 
> > When the ADR_ACK_CNT reaches the EU868_ADR_ACK_LIMIT(64), **the txPower is set to EU868_MAX_TX_POWER(0)**.
> 
> Why is the device doing this? I'm not aware that this is in line with the LoRaWAN specifications. If you believe it is, could you refer to the section describing this?

It´s on Semtech LoRaMac-node implementation:

https://github.com/Lora-net/LoRaMac-node/blob/83b6086f33aaef251820b91e4ac2b35c9c7b4a91/src/mac/region/RegionEU868.c#L504

brocaar, 2019-04-11:
I have created this issue: https://github.com/Lora-net/LoRaMac-node/issues/723

rayman18, 2019-04-15:
> 
> 
> I have created this issue: [Lora-net/LoRaMac-node#723](https://github.com/Lora-net/LoRaMac-node/issues/723)

From their answer: 

> In case the network server receives the uplink and then the end-device receives a downlink. Then the network server ADR algorithm will anyway send a LinkAdrReq mac command as it will notice that something changed. (may take a few uplinks depending on the ADR algorithm being used)

@brocaar what do you think about this?