brocaar/chirpstack-network-server #390: Build from source problem

Silverlay, 2019-04-11:
### Is this a bug or a feature request?
Seems like a bug

### What did you expect?
Make complete

### What happened?
Got
root@SPH-LWN02 [ loraserver ]# make serve
Generating static files
Compiling source
go build  -ldflags "-s -w -X main.version=2.6.1-5-gd454003" -o build/loraserver cmd/loraserver/main.go
# github.com/brocaar/loraserver/internal/downlink/multicast
internal/downlink/multicast/min_gw_set.go:144:39: cannot use deviceGatewayEdge literal (type deviceGatewayEdge) as type graph.WeightedEdge in argument to g.SetWeightedEdge:
        deviceGatewayEdge does not implement graph.WeightedEdge (missing ReversedEdge method)
make: *** [build] Error 2

### What version are your using?
latest

### How can your issue be reproduced?
On CentOS Linux used comman by command as follows

#install prerequisites
yum -y install git alien autoconf automake libtool curl make g++ unzip
yum -y groupinstall "Development Tools"
yum -y install kernel-devel kernel-headers

#install go
curl -O https://dl.google.com/go/go1.11.3.linux-amd64.tar.gz
sudo tar -C /usr/local -xvzf go1.11.3.linux-amd64.tar.gz
export PATH=$PATH:/usr/local/go/bin
export GOPATH=$HOME/go
export GOBIN=$GOPATH/bin
export PATH=$PATH:/usr/local/go/bin:$GOBIN
source $HOME/.profile
source $HOME/.bashrc

#install protocol buffers (3.7.1)
cd ~/home/noradius
wget https://github.com/protocolbuffers/protobuf/releases/download/v3.7.1/protobuf-all-3.7.1.tar.gz
tar xvfz protobuf-all-3.7.1.tar.gz
cd protobuf-3.7.1/
./configure && make install

#install Go protobuf
cd $GOPATH
go get -u -v github.com/golang/protobuf/proto
go get -u -v github.com/golang/protobuf/protoc-gen-go

#install Go requirements
cd $GOPATH
go get -u -v github.com/jteeuwen/go-bindata
go install github.com/jteeuwen/go-bindata
go get -u -v golang.org/x/lint/golint
go get -u -v golang.org/x/tools/cmd/stringer
go get -u -v github.com/golang/protobuf/protoc-gen-go
go get -u -v github.com/grpc-ecosystem/grpc-gateway/protoc-gen-swagger
go get -u -v github.com/grpc-ecosystem/grpc-gateway/protoc-gen-grpc-gateway
go get -u -v github.com/elazarl/go-bindata-assetfs/go-bindata-assetfs
go get -u -v github.com/jteeuwen/go-bindata/go-bindata
go get -u -v github.com/goreleaser/goreleaser
go get -u -v github.com/goreleaser/nfpm
go get -u -v cloud.google.com/go/
go get -u -v github.com/NickBall/go-aes-key-wrap
go get -u -v github.com/eclipse/paho.mqtt.golang
go get -u -v github.com/gofrs/uuid
go get -u -v github.com/gomodule/redigo/redis
go get -u -v github.com/googleapis/gax-go/v2
go get -u -v github.com/rubenv/sql-migrate
go get -u -v github.com/sirupsen/logrus
go get -u -v github.com/spf13/cobra
go get -u -v github.com/spf13/viper
go get -u -v go.opencensus.io/plugin/ocgrpc
go get -u -v go.opencensus.io/stats
go get -u -v go.opencensus.io/stats/view
go get -u -v go.opencensus.io/tag
go get -u -v gonum.org/v1/gonum/graph
go get -u -v gonum.org/v1/gonum/graph/path
go get -u -v gonum.org/v1/gonum/graph/simple
go get -u -v google.golang.org/api/iterator
go get -u -v google.golang.org/api/option
go get -u -v google.golang.org/api/support/bundler
go get -u -v google.golang.org/api/transport
go get -u -v github.com/grpc-ecosystem/go-grpc-middleware
go get -u -v github.com/jmoiron/sqlx
go get -u -v github.com/lib/pq
go get -u -v github.com/smartystreets/goconvey/convey
go get -u -v github.com/stretchr/testify/require

#install brocaar local requirements
go get -u -v github.com/brocaar/lorawan

#install brocaar/loraserver
go get -u -v github.com/brocaar/loraserver
cd $GOPATH/src/github.com/brocaar/loraserver
make dev-requirements
make serve

### Could you share your log output?
Yup, on make serve is upper. On make test as follows

root@SPH-LWN02 [ loraserver ]# make test
Generating static files
Running tests
# github.com/brocaar/loraserver/internal/downlink/multicast
internal/downlink/multicast/min_gw_set.go:144:39: cannot use deviceGatewayEdge literal (type deviceGatewayEdge) as type graph.WeightedEdge in argument to g.SetWeightedEdge:
        deviceGatewayEdge does not implement graph.WeightedEdge (missing ReversedEdge method)
# github.com/brocaar/loraserver/internal/downlink/multicast [github.com/brocaar/loraserver/internal/downlink/multicast.test]
internal/downlink/multicast/min_gw_set.go:144:39: cannot use deviceGatewayEdge literal (type deviceGatewayEdge) as type graph.WeightedEdge in argument to g.SetWeightedEdge:
        deviceGatewayEdge does not implement graph.WeightedEdge (missing ReversedEdge method)
make: *** [test] Error 2

Silverlay, 2019-04-11:
nvm, messed up with go libs :(

brocaar, 2019-04-11:
Ok. anyway I recommend using the Docker environment for building. This environment is also used for the CI pipeline :-)

closed 2019-04-11 by brocaar