brocaar/chirpstack-network-server #392: Set channel plan per device

lancepitka, 2019-05-09:
### Is this a bug or a feature request?

Feature Request

### What did you expect?

For US 915 networks with some gateways with 8 channels and some with 16 channels, having one channel plan for all devices on the network doesn't work very well. It would be great if we could override the default channel plan for certain devices. That way, the default plan could be to use all 16 channels by default, but override the channel plan to only 8 channels for devices using an 8 channel gateway for example.

### What happened?

### What version are your using?

### How can your issue be reproduced?

### Could you share your log output?

brocaar, 2019-05-14:
I would recommend setting up two LoRa Server instances one for a 8 channel US915 network, an other for a 16 channel US915 network.

Ideally the end-user setting up devices should not be concerned with the network specific settings (like active channels). Giving the end-user the power of selecting which channels to use could cripple your network.

lancepitka, 2019-05-14:
Setting up two different network servers wouldn't work for my case. I have many devices that move around between 8 and 16 channel gateways. Unless there is a way to setup some sort of roaming between my two network servers, I don't think it would work.

brocaar, 2019-05-14:
Potentially you could forward all data to a single MQTT broker and let both LoRa Server instances use the uplink data from all 16 ch + 8 ch gateways.

Make sure that each LoRa Server instance has its own Redis and PostgreSQL database. So they share the same gateways, but not the device states.

lancepitka, 2019-05-14:
That's a good idea. I think that should work. I will give it a try. Thanks!

Is there a way to disable the gateway channel reconfiguration packet on the loraserver? I can ignore it on the gateways, but having that packet sent from each network server uses that much more data. I don't know how much data this actually uses - maybe it is negligible?

brocaar, 2019-05-14:
Just not assign a gateway profile :-) The only thing it is used for is sending the config to the LoRa Gateway Bridge.

lancepitka, 2019-05-14:
I didn't realize that! Thanks for the help

brocaar, 2019-06-05:
Can this issue be closed?

lancepitka, 2019-06-06:
Yes it can. Thanks!

From: Orne Brocaar
Sent: Wednesday, June 5, 2019 7:00 AM
To: brocaar/loraserver
Cc: lancepitka; Author
Subject: Re: [brocaar/loraserver] Set channel plan per device (#392)

Can this issue be closed?

closed 2019-06-06 by lancepitka