brocaar/chirpstack-network-server #395: For MAC Version 1.1, devices configured as Class C join as Class A

AnthonySorvariOATI, 2019-05-20:
### Is this a bug or a feature request?

Bug

### What did you expect?

In LoRaWAN 1.0.2, device class definition is done out-of-band, hence the “Supports Class C” setting in LoRa Server device profiles. Given a device profile with MAC Version = 1.0.2 and Supports Class C = true, the device will be treated by LoRa Server as a Class C device. In LoRaWAN 1.1, two new MAC commands “DeviceModeInd” and “DeviceModeConf” were added to allow devices to transition between Class A and Class C. However, there is no mention made in the spec of requiring the device to send DeviceModeInd as part of the join process. We expect that true Class C devices (as opposed to Class A devices which only temporarily switch to Class C for e.g. FUOTA) will behave the same with MAC Version 1.1 as with 1.0.2. Our intention is to configure the devices as Class C out-of-band even if they do not send DeviceModeInd commands.

### What happened?

Given a device profile with MAC Version = 1.1 and Supports Class C = true, the device is treated by LoRa Server as a Class A device upon joining.

### What version are you using?

loraserver: 2.6.1  
lora-app-server: 2.6.0  
lora-gateway-bridge: 2.7.1

### How can your issue be reproduced?

1. Create a Device profile with LoRaWAN MAC version = 1.1.0, LoRaWAN Regional Parameters revision = B, and Class-C enabled.
1. Create electric meter Device with this Device profile. 
1. Once Device successfully joins Loraserver, verified the Device in the Device table Mode column value in loraserver_ns database to be “A”. 
1. Send a downlink command with application.  
   1. Downlink command message sits in Loraserver Device queue until an uplink from the Device occurs, which is how a class A Device behaves. 
   1. Device live Lorawan frames show uplink and then downlink message.
   1. Device responds as expected per command. 

When the Device profile has LoRaWAN MAC version = 1.0.2, the Device table Mode column value in loraserver_ns is set to “C” instead, and downlink command message gets processed and removed from the Loraserver Device queue within seconds. Device live Lorawan frames show downlink with no uplink first.

### Could you share your log output?

Device live Lorawan frames shows downlink occurring after uplink when Mode is “A”.

[image]

Device live Lorawan frames shows only downlink when Mode is “C”.

[image]

brocaar, 2019-06-24:
I have been asking around at the LoRa Alliance meeting two weeks ago, and you are right. A Class-C device can join the network as a Class-C device directly.

brocaar, 2019-06-26:
This has been fixed in the above commit. I will also make an update to LoRa App Server so that the Class-C time-out can be configured even when the device does not support Class-C initially, but changes state later by sending a `DeviceModeInd` mac-command.

commit b5339b95 (2019-06-26): Revert LW1.1 Class-C device always joins as Class-A.

The LoRaWAN 1.1 specification does not specify that a Class-C device
must always signal it is Class-C capable using the DeviceModeInd
mac-command.

Closes #395.
closed 2019-06-26 by brocaar