brocaar/chirpstack-network-server #397: TXParamSetupReq MAC commands not being sent (AS923)

stevieraysean, 2019-05-29:
### Is this a bug or a feature request?
bug  (possibly not implemented feature)

### What did you expect?
TXParamSetupReq should be sent to a device joining an AS923 or AU915 network.

### What happened?
No matter what Dwell_TIme_400ms or repeater_compatible variables set in loraserver.toml. no TXParamSetupReq MAC command is sent to a joining device on AS923. dwell-time remains on always.

### What version are your using?
loraserver 2.8.2

### How can your issue be reproduced?
Setting channel plan to AS923, get an end device to join. no TXParamSetupReq sent, dwell-time never disabled on end device 

### Could you share your log output?
only mac Req commands sent:
journalctl -f -n 20 -u loraserver | grep Req

May 29 15:56:01 sean-ThinkPad-L470 loraserver[8356]: time="2019-05-29T15:56:01+12:00" level=info msg="pending mac-command block set" cid=NewChannelReq commands=1 dev_eui=70b3d5fbf000e0ce
May 29 15:56:01 sean-ThinkPad-L470 loraserver[8356]: time="2019-05-29T15:56:01+12:00" level=info msg="pending mac-command block set" cid=DevStatusReq commands=1 dev_eui=70b3d5fbf000e0ce
May 29 15:56:04 sean-ThinkPad-L470 loraserver[8356]: time="2019-05-29T15:56:04+12:00" level=info msg="pending mac-command deleted" cid=NewChannelReq dev_eui=70b3d5fbf000e0ce
May 29 15:56:04 sean-ThinkPad-L470 loraserver[8356]: time="2019-05-29T15:56:04+12:00" level=info msg="pending mac-command deleted" cid=DevStatusReq dev_eui=70b3d5fbf000e0ce
May 29 15:56:04 sean-ThinkPad-L470 loraserver[8356]: time="2019-05-29T15:56:04+12:00" level=info msg="pending mac-command block set" cid=LinkADRReq commands=1 dev_eui=70b3d5fbf000e0ce

brocaar, 2019-07-02:
You are right, this mac-command hasn't been implemented yet. I'm happy to implement this. When I have a test-version, would you be able to validate this for AS923?

stevieraysean, 2019-07-02:
great!, happy to verify for AS923..

brocaar, 2019-07-03:
Great. This is an example how this will be reflected in the configuration (WIP):

```toml
  # Enforce 400ms dwell time.
  #
  # Some regions require the configuration of the dwell time, which will
  # limit the time-on-air to 400ms. Please refer to the LoRaWAN Regional
  # Parameters specification for more information.
  #
  # When configured and required in the configured region, LoRa Server will
  # use the TxParamSetup mac-command to communicate this to the devices.
  uplink_dwell_time_400ms={{ .NetworkServer.Band.UplinkDwellTime400ms }}
  downlink_dwell_time_400ms={{ .NetworkServer.Band.DownlinkDwellTime400ms }}

  # Uplink max. EIRP.
  #
  # This defines the maximum allowed device EIRP which must be configured
  # for some regions. Please refer the LoRaWAN Regional Parameters specification
  # for more information. Set this to -1 to use the default value for this
  # region.
  #
  # When required in the configured region, LoRa Server will use the
  # TxParamSetup mac-command to communicate this to the devices.
  # For regions where the TxParamSetup mac-command is not implemented, this
  # setting is ignored.
  uplink_max_eirp={{ .NetworkServer.Band.DeviceMaxEIRP }}
```

The mac-command supports setting both the uplink and downlink dwell time, so I needed to split this. Also the mac-command sets the device max. EIRP, so I have to add this setting too. Note that when this setting is also configured in the device-profile, I believe the smallest value must be used to be compatible with local regulations.

* You don't want users to configure a higher EIRP through the device-profile (as this might violate regulations)
* When you configure e.g. 20 but the device supported max EIRP is 14, then 14 must be communicated to the device

brocaar, 2019-07-08:
I believe https://github.com/brocaar/loraserver/compare/tx_param_setup should implement the TXParamSetup mac-command.

You could compile this branch yourself, but I have also created test binaries and .deb packages for you: https://www.dropbox.com/sh/9w9wnfig368yxin/AAA7rvLbOJ_I-W9wQ7pHgBVVa?dl=0

Looking forward to your feedback! Note that you will need to test this together with LoRa Gateway Bridge and LoRa App Server v3.

You probably want to set the following variables (see full config template below):

* uplink_dwell_time_400ms
* downlink_dwell_time_400ms
* uplink_max_eirp

Configuration template:

```toml
[general]
# Log level
#
# debug=5, info=4, warning=3, error=2, fatal=1, panic=0
log_level=4

# PostgreSQL settings.
#
# Please note that PostgreSQL 9.5+ is required.
[postgresql]
# PostgreSQL dsn (e.g.: postgres://user:password@hostname/database?sslmode=disable).
#
# Besides using an URL (e.g. 'postgres://user:password@hostname/database?sslmode=disable')
# it is also possible to use the following format:
# 'user=loraserver dbname=loraserver sslmode=disable'.
#
# The following connection parameters are supported:
#
# * dbname - The name of the database to connect to
# * user - The user to sign in as
# * password - The user's password
# * host - The host to connect to. Values that start with / are for unix domain sockets. (default is localhost)
# * port - The port to bind to. (default is 5432)
# * sslmode - Whether or not to use SSL (default is require, this is not the default for libpq)
# * fallback_application_name - An application_name to fall back to if one isn't provided.
# * connect_timeout - Maximum wait for connection, in seconds. Zero or not specified means wait indefinitely.
# * sslcert - Cert file location. The file must contain PEM encoded data.
# * sslkey - Key file location. The file must contain PEM encoded data.
# * sslrootcert - The location of the root certificate file. The file must contain PEM encoded data.
#
# Valid values for sslmode are:
#
# * disable - No SSL
# * require - Always SSL (skip verification)
# * verify-ca - Always SSL (verify that the certificate presented by the server was signed by a trusted CA)
# * verify-full - Always SSL (verify that the certification presented by the server was signed by a trusted CA and the server host name matches the one in the certificate)
dsn="postgres://localhost/loraserver_ns?sslmode=disable"

# Automatically apply database migrations.
#
# It is possible to apply the database-migrations by hand
# (see https://github.com/brocaar/loraserver/tree/master/migrations)
# or let LoRa App Server migrate to the latest state automatically, by using
# this setting. Make sure that you always make a backup when upgrading Lora
# App Server and / or applying migrations.
automigrate=true

# Redis settings
#
# Please note that Redis 2.6.0+ is required.
[redis]
# Redis url (e.g. redis://user:password@hostname/0)
#
# For more information about the Redis URL format, see:
# https://www.iana.org/assignments/uri-schemes/prov/redis
[... 520 lines cut ...]
  # Please refer to the LoRaWAN Backend Interface specification
  # 'Key Transport Security' section for more information.
  #
  # Example (the [[join_server.kek.set]] can be repeated):
  # [[join_server.kek.set]]
  # # KEK label.
  # label="000000"

  # # Key Encryption Key.
  # kek="01020304050607080102030405060708"

  # Network-controller configuration.
  [network_controller]
  # hostname:port of the network-controller api server (optional)
  server=""

  # ca certificate used by the network-controller client (optional)
  ca_cert=""

  # tls certificate used by the network-controller client (optional)
  tls_cert=""

  # tls key used by the network-controller client (optional)
  tls_key=""
```

stevieraysean, 2019-07-09:
wow, thanks. 

I'm trying this out now, seem to be getting an error right after the join device-status request

`Jul 09 05:09:46 ip-172-30-2-95 loraserver[32440]: time="2019-07-09T05:09:46Z" level=info msg="adr request added to mac-command queue" dev_eui=70b3d5fbf000e0ce dr=2 nb_trans=1 req_dr=5 req_nb_trans=1 req_tx_power_idx=2 tx_power=0`

`Jul 09 05:09:46 ip-172-30-2-95 loraserver[32440]: time="2019-07-09T05:09:46Z" level=info msg="requesting device-status" dev_eui=70b3d5fbf000e0ce`

`Jul 09 05:09:46 ip-172-30-2-95 loraserver[32440]: time="2019-07-09T05:09:46Z" level=error msg="processing uplink frame error" data_base64="QEC8dQGAAAAB7jKqXIV/a+awdYqJIeM=" error="run uplink response flow error: get mac-command block size error: marshal binary error: lorawan: invalid MaxEIRP value"`

not sure whether this is a loraserver issue, or in our end devices. 
have MaxEIRP set to 16 in device profile, and -1 on loraserver ( the default for AS923 is 16 i believe)

have tried other values, but the error persists...

thanks again.

brocaar, 2019-07-09:
Let me check, this could be an error at my side. Thanks for sharing the logs :-)

brocaar, 2019-07-09:
@stevieraysean could you try again? I have created new snapshot binaries (SNAPSHOT-2a66b51) and have put these in the same Dropbox share.

stevieraysean, 2019-07-09:
@brocaar fantastic!
seems to be all working as expected now.

brocaar, 2019-07-10:
Great! You're right about the downlink, there was a typo to map the config-key to the struct. This has been fixed and I'll merge this into master.

Thanks again for testing :+1:

commit 76999ae9 (2019-07-10): Implement TXParamSetup req/ans mac-commands.

This is used to configure the uplink / downlink dwell time of the device
as well the max. allowed EIRP of the device.

Closes #397.
closed 2019-07-10 by brocaar