brocaar/chirpstack-network-server #398: limit the number of redis connections get by lorasever

orientlu, 2019-05-30:
### Is this a bug or a feature request?
i think is a bug

### What did you expect?
i try to fix and  had subminted a pr

### What happened?
when i do a perf test ,i found loraserver print so much error log which say redis connection timeout, and i found loraserver get more than 9990 connection(max is 10000).
### What version are your using?

### How can your issue be reproduced?

### Could you share your log output?

orientlu, 2019-05-30:
and i suggest postgsql should do the limit too， because i encountered the situatioon which as total did not work because ns get too much psql client connections。

brocaar, 2019-06-05:
:+: This has been fixed. I've updated your pull-request to make this a config option :-)

closed 2019-06-05 by brocaar

orientlu, 2019-06-11:
> :+: This has been fixed. I've updated your pull-request to make this a config option :-)
and postgsql