brocaar/chirpstack-network-server #400: Internal Server Error after getting the queue

DHintCo, 2019-06-05:
### Is this a bug or a feature request?
A Bug

### What happened?
The device transmits uplink but does not receive downlink after changing the server version

### What version are your using?
In the state in which the nodes are working, I have used version 2.6.0 and the version in which it does not work is 2.6.1.

### How can your issue be reproduced?
I am trying to get the device queue through the API

### Could you share your log output?
Maybe it is important to know, that the DevAddr consists of numbers only

[image]

brocaar, 2019-06-05:
That is strange, since nothing changed with regards to the database functions between v2.6.0 and v2.6.1: https://github.com/brocaar/loraserver/compare/v2.6.0...v2.6.1. 

> I am trying to get the device queue through the API

* I make the assumption here that you mean the `/api/devices/{dev_eui}/queue` endpoint?
* What is the LoRa App Server version?

DHintCo, 2019-06-05:
Ah, sorry I wrote into the wrong repository...
This issue should be moved to the lora-app-server.

Now to your questions:
- Yes, you are right. I mean the `/api/devices/{dev_eui}/queue` endpoint

- Both LoRaserver and the Applicationserver were deployed in version 2.6.0 and both were changed to 2.6.1.

DHintCo, 2019-06-19:
Are there new information?

brocaar, 2019-07-02:
I have not been able to reproduce this issue.

closed 2019-11-13 by brocaar