brocaar/chirpstack-network-server #401: loraserver ignoring packets from multiple Rx paths

KarthikSubramanian2, 2019-06-10:
### Is this a bug or a feature request?
bug

### What did you expect?
Even though there are multiple packets generated for the same uplink through one gateway, loraserver should dedup them and not lock based on just the first packet.

### What happened?
Since the lock is only on the first packet with the key `lora:ns:uplink:lock:<GatewayID>:<PhyPayload>"`, there is a chance of the original packet with a better signal strength getting dropped which causes a wrong downlink and a bunch of retry packets(which also get dropped with the FC validation enabled causing the sensor to rejoin).

### What version are your using?
2.6.1

### How can your issue be reproduced?
Simulate two Rx packets with the same gateway ID and different RSSIs and different channels. 

### Could you share your log output?
```
"SET" "lora:ns:uplink:lock:0000000000000002:80ab47c80000000001f6a5bedb10214997c409a9cd" "lock" "PX" "500" "NX"
```

KarthikSubramanian2, 2019-06-10:
If we can add `Channel` as part of the lock key to eliminate the multiple subscription issue, then the duplicate packets due to multiple Rx paths might go to Redis and then the sorting algorithm would pick up the right packet.

brocaar, 2019-06-24:
How close is your device to the gateway? Usually this happens only when your device is really close.

KarthikSubramanian2, 2019-06-24:
Thanks for your reply. You are spot on. We have observed this issue in two different scenarios. 
1) As you mentioned, when the device signal strength is too strong (close to the gateway)
2) Also when the gateway is in a highly reflective environment(in our cases with proximity to multiple small metal containers) 

For a small rollout, moving the sensor or gateway would probably solve it. But when we have 100s of sensors and multiple gateways in a constrained space, this issue is really difficult to avoid. 

In those scenarios, it would be great if our software solution handles it and not be concerned about the gateway and sensor proximity. I also like that loraserver is built in a way that the fix should be easy to incorporate as well. 

Curious to hear your thoughts.

KarthikSubramanian2, 2019-07-01:
@brocaar I can test it out and create a PR if everything looks good.

brocaar, 2019-07-02:
@KarthikSubramanian2 yes please create a pull-request for this :+1: I think the best way would be to add the Frequency to the key, not the Channel (number). E.g.

```
"SET" "lora:ns:uplink:lock:0000000000000002:868100000:80ab47c80000000001f6a5bedb10214997c409a9cd" "lock" "PX" "500" "NX"
```

PR #409 (2019-07-05): Adding frequency as part of the lock key
This would prevent the duplicate uplink packets due to multiple Rx paths from getting dropped.
Fixes #401 

Output from `redis-cli monitor`

_Before change:_
```
1562097425.235034 [0 10.112.147.63:51622] "SET" "lora:ns:uplink:lock:d34f34d34d346b4d:8092a08901040b000306030002fc9f10d8a052fd389e" "lock" "PX" "500" "NX"
```

_After change:_
```
1562098084.656540 [0 10.112.147.63:53561] "SET" "lora:ns:uplink:lock:d34f34d34d346b4d:868100000:8092a08901040b000306030002fc9f10d8a052fd389e" "lock" "PX" "500" "NX"
```
commit 82ebb37b (2019-07-05): Add frequency as part of the lock key (#409)

Closes #401.
closed 2019-07-05 by brocaar