brocaar/chirpstack-network-server #408: Node Activation without Uplink from Node to Application

andhir0012, 2019-07-01:
### Is this a bug or a feature request?
I think it is a feature based on the normal operation.

### What did you expect?
Activation of a device should be completed if the device joins the LNS.
If the device sends an uplink frame for configuration purpose the activation should be done.

### What happened?
Class-A device is a commercial product.
Device Activation is not completed.
I can start a so called configuration mode on a node. In this configuration mode the node sends an uplink WITHOUT payload data for 10 Minutes. This Uplink frame do not lead to activation in Application Server (I think because there is no payload within this frame).

### What version are your using?
Loraserver version=3.0.2
Applicationserver version=3.0.0

### How can your issue be reproduced?
Always the Node Joins the LNS

### Could you share your log output?

time="2019-07-01T12:13:36Z" level=info msg="gateway/mqtt: uplink frame received"
time="2019-07-01T12:13:36Z" level=info msg="rx info sent to network-controller" dev_eui=70b3d5e75e006862
time="2019-07-01T12:13:36Z" level=info msg="device gateway rx-info meta-data saved" dev_eui=70b3d5e75e006862
time="2019-07-01T12:13:36Z" level=info msg="device-session saved" dev_addr=0075c057 dev_eui=70b3d5e75e006862

No log from Application Server.

Live LoraWan Frame where you can see the null payload
devAddr:"0075c057"
        adr:true
        adrAckReq:false
        ack:false
        fPending:false
        classB:false
        fCnt:23
        fOpts:null
    fPort:null
    frmPayload:null
mic:"ae5a56b2"

brocaar, 2019-07-09:
What I'm planning to do is **always** forward an uplink event to the AS, even when FPort is empty or FPort = 0. With this LoRa Server will not wait with signaling the security-context change to the AS.

I will then also update LoRa App Server to not generate `.../up` events in case of FPort = 0 to avoid that the end-application will receive empty payloads in such events.

closed 2019-07-09 by brocaar

andhir0012, 2019-07-09:
Thank you
for doing this changes. I will give you feedback about the functionality.

You do a very good job in this project.

Greetings

commit 63613c0e (2019-07-09): Allways forward uplink data request (even on FPort= nil | 0).

Even when no application-payload is sent, this can still provide
valuable information to the end-application (e.g. data-rate, RX
attributes, the fact that the device is 'alive').

Closes #408.