brocaar/chirpstack-network-server #420: Introduce "Congestion Control" for MACCommands

JanZimmer, 2019-08-19:
### Is this a bug or a feature request?
Feature
### What did you expect?
Technically, the LoRaWAN specification makes no assumptions on how to handle non-conforming devices. However, we have seen LoRaWAN sensors, that do not reply to e.g. RXParamSetupReq MAC Commands, which leads to a non stopping flood of downlink MAC Commands. Even though, this is not the "fault" of the LNS, maybe their should be something mitigating the spamming of the downlink.
### What version are your using?
all
### How can your issue be reproduced?
yes

brocaar, 2019-09-04:
I understand your issue, but I think it is also a good idea (if you haven't already done) to report this to the device vendors to make sure they are aware of these issues in their firmware.

JanZimmer, 2019-09-04:
I totally agree and this is of cause the strategy that we are currently using. However, to get a production ready setup, some kind of fault tolerance might be a good feature, anyway.

commit 8e1ad826 (2019-11-26): Implement max. mac-command error counter.

When a mac-command is nACKed for more than the configured value, then the
ChirpStack Network Server will stop sending this mac-command to the device.
This setting prevents that the Network Server will keep sending mac-commands
on every downlink in case of a malfunctioning device.

Closes #420.
closed 2019-11-26 by brocaar