brocaar/chirpstack-network-server #421: Introduce the possibility to stick to the device Profile

JanZimmer, 2019-08-19:
### Is this a bug or a feature request?
Feature
### What did you expect?
It would be handy, to make it possible to let the LNS stick to the settings in the device profile instead of trying to overwrite them with its global setup (the one done in the config file).

This would add more felxibility on efficient planning of network parameters for a heterogenous setup.

Just a flag to tick would be fine.

brocaar, 2019-08-21:
But since the device-profile can be created / managed by the end-user, this will enable the end-user to define the used radio parameters (like RX2 data-rate and frequency). I'm not sure if you want to give this power to your end-users.

JanZimmer, 2019-08-21:
You're right. Maybe just allow admins to see/set this flag?

JanZimmer, 2019-08-21:
Or one could connect such feature to the capabilities of ServiceProfile, so that the user is always limited by it, regardless of what she configures in the DeviceProfile.

brocaar, 2019-08-21:
What would be needed to put the "efficient planning of network parameters for a heterogenous setup" in the `loraserver.toml` configuration?

We could put something in the service- and device- profile, but ideally this would be the concern of the LoRa Server administrator, not the device owner. The LoRa Server administrator has knowledge about the network, gateways, latency between the NS and gateways, capacity etc...

To start with, what exact parameters are you referring to with "settings in the device profile"?

> This would add more felxibility on efficient planning of network parameters for a heterogenous setup.

Could you describe a couple of use-cases for which you want to use this feature?

brocaar, 2020-01-24:
I'm closing this issue.

closed 2020-01-24 by brocaar