brocaar/chirpstack-network-server #422: Add more options on downlink behavior

JanZimmer, 2019-08-19:
### Is this a bug or a feature request?
Feature
### What did you expect?
It would be nice, if there would be the option to configure the downlink behavior a bit more fine-grained.

Imagine the following example:
A static node sends uplink on SF12 to the gateway. According to the Link-Budget (consisting of the SF and the TX Power) of such node (at least for Europe), there is no reason to send downlink in RX1 with SF12, as the Link-Budget of the RX2 downlink channel (in the P-Band) has better duty-cycle (10% vs. 1% for RX1 Bands) and you are allowed to send with such a high TX-Power that even SF9 has a better downlink Link-Budget then the original uplink message.

Lets say a clever way of using this would be to extend the `rx_window`configuration behavior as follows:
1. introduce a new option for `rx_window` (say `3`) to enable the new behavior
2. introduce a new configuration (lets say `rx2_only_dr`)
3. if `DR of uplink > rx2_only_dr` only schedule RX2 downlink, else schedule RX1 and RX2

As the cherry on top, it would be great to make this configurable in the device profile.

brocaar, 2019-08-21:
I think this is a good one. Would you like to contribute / help implement this feature?

I think it would also require to update this function (this is for the EU band, but this might apply to other bands too):

https://github.com/brocaar/lorawan/blob/master/band/band_eu863_870.go#L29

What would be needed is to define the frequency ranges + the TX power that is allowed for that range.

> As the cherry on top, it would be great to make this configurable in the device profile.

Let's not include this for now. See also my comment to your other issue. Personally I do like to separate the network and the end-user part.

JanZimmer, 2019-08-26:
So I started implementing the feature. However, I'm not quite familiar with go yet. Just have a look at https://github.com/JanZimmer/loraserver/tree/feature-rx2onlydr as a discussion base. I'm not sure, how "good" the automatic link budget calculation will work and if it is a good idea to do this in the first place.

In https://github.com/JanZimmer/lorawan/tree/fix_eu_868_tx_power , I also tried adopting the TX power for EU868.

JanZimmer, 2019-09-04:
@brocaar did you have a chance to have a look at the proposed changes?

brocaar, 2019-12-10:
This has been merged :)

closed 2019-12-10 by brocaar