brocaar/chirpstack-network-server #425: Makefile:46: recipe for target 'statics' failed

Sadashiv97, 2019-08-28:
### Is this a bug or a feature request?

### What did you expect?

### What happened?

### What version are your using?

### How can your issue be reproduced?

### Could you share yo
[image]
ur log output?

brocaar, 2019-08-28:
The recommended way is to use the provided Docker image for compiling the source, please see https://www.loraserver.io/loraserver/community/source/ for instructions and commands to execute.

closed 2019-08-28 by brocaar

mittal-pooja, 2019-09-02:
> This issue is present but closed but now i am feeling this difficult to debug this problem though it is a small error i think so
> please guide me to debug this
> [image]
> 
> I installed **GO** language using command
> **sudo snap install go --classic**
> **go version** =>go version go1.12.9 linux/amd64
> then cloning the git repository **git clone https://github.com/brocaar/loraserver.git**
> **cd loraserver**
> **sudo make dev-requirements**
> **sudo make**
> I am getting the **go-bindata** executable file not found in $PATH
> so i did **go get -u github.com/jteeuwen/go-bindata/...**
> as @brocaar said
> not worked for me
> then i found the bin directory where dev-requirements were installed i did
> **export PATH=$PATH:$GOPATH/bin**
> 
> giving same Error
> 
> help me !!
> 
> **ignore my english please**

Hey @brocaar !
I am also getting the same issue. There seems to be some kind of problem in migrations.go file. 

Previously, I compiled the code on `Ubuntu 18.04.3 LTS`(without Docker), 4 times, and it compiled successfully, but right now, I am trying to compile the code on `Ubuntu 16.04.6 LTS,` it is giving the same error as above, when I try to `make build`. 
Is there some kind of dependency on Ubuntu Version? 

The `go-bindata binary` is already present in my `$GOPATH/bin` but it still can't find it in the $PATH. **_Kindly suggest a solution to this(without using Docker)._**

error: ```
Generating static files
internal/migrations/migrations.go:1: running "go-bindata": exec: "go-bindata": executable file not found in $PATH
Makefile:46: recipe for target 'statics' failed
make: *** [statics] Error 1
```
`GOPATH="/home/network-server"`

`$PATH = -bash: /home/network-server/bin:/usr/local/go/bin:/home/network-server/bin:/usr/local/go/bin/bin:/home/network-server/bin:/usr/local/go/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/snap/bin: No such file or directory
`