brocaar/chirpstack-network-server #427: Already closed issue, but it does not provide any solution to it! "go-bindata executable not found in $PATH"

mittal-pooja, 2019-09-02:
> This issue is present but closed but now i am feeling this difficult to debug this problem though it is a small error i think so
> please guide me to debug this
> [image]
> 
> I installed **GO** language using command
> **sudo snap install go --classic**
> **go version** =>go version go1.12.9 linux/amd64
> then cloning the git repository **git clone https://github.com/brocaar/loraserver.git**
> **cd loraserver**
> **sudo make dev-requirements**
> **sudo make**
> I am getting the **go-bindata** executable file not found in $PATH
> so i did **go get -u github.com/jteeuwen/go-bindata/...**
> as @brocaar said
> not worked for me
> then i found the bin directory where dev-requirements were installed i did
> **export PATH=$PATH:$GOPATH/bin**
> 
> giving same Error
> 
> help me !!
> 
> **ignore my english please**

Hey @brocaar !
I am also getting the same issue. There seems to be some kind of problem in migrations.go file. 

Previously, I compiled the code on `Ubuntu 18.04.3 LTS`(without Docker), 4 times, and it compiled successfully, but right now, I am trying to compile the code on `Ubuntu 16.04.6 LTS,` it is giving the same error as above, when I try to `make build`. 
Is there some kind of dependency on Ubuntu Version? 

The `go-bindata binary` is already present in my `$GOPATH/bin` but it still can't find it in the $PATH. **_Kindly suggest a solution to this(without using Docker)._**

error:   ```Generating static files
internal/migrations/migrations.go:1: running "go-bindata": exec: "go-bindata": executable file not found in $PATH
Makefile:46: recipe for target 'statics' failed
make: *** [statics] Error 1```

`GOPATH="/home/network-server"`

`$PATH = -bash: /home/network-server/bin:/usr/local/go/bin:/home/network-server/bin:/usr/local/go/bin/bin:/home/network-server/bin:/usr/local/go/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/snap/bin: No such file or directory
`

_Originally posted by @mittal-pooja in https://github.com/brocaar/loraserver/issues/425#issuecomment-527009034_

brocaar, 2019-09-02:
> I am also getting the same issue. There seems to be some kind of problem in migrations.go file.

There is no issue with the `migrations.go` file, the error you're seeing is related to an incorrect `PATH` variable and the `go-bindata` can not be found.

**Please use the provided Docker development environment for building**. Then you will have exactly the same environment as I'm using for building the releases with all the variables correctly configured.

I can not support non-Docker environments as I do not have any visibility on how you have setup your machine.

closed 2019-09-02 by brocaar

mittal-pooja, 2019-09-03:
Hey, @brocaar , thank you so much for your answer!

> Previously, I compiled the code on `Ubuntu 18.04.3 LTS`(without Docker), 4 times, and it compiled successfully, but right now, I am trying to compile the code on `Ubuntu 16.04.6 LTS,` it is giving [...]

As, it turns out, if you're compiling the code without docker, then there is some kind of dependency on Ubuntu Version [I am not aware of the reason behind this, though]. 
Previously, I was trying to compile the source code on `Ubuntu 16.04.6 LTS`, where this error persisted, even after the correct `PATH `definition and also making sure that the `binary` is indeed present in the `PATH`.
But, after I upgraded to `Ubuntu 18.04.3 LTS`, and following the same steps as before, I didn't get this error. The code was compiled as well built smoothly.