brocaar/chirpstack-network-server #430: Time synchronize and batch upload

mttbx, 2019-09-13:
I want to know if there's some methods to synchronize my device time (know the real world time) . And I also want to know if there's some methods to batch the messages before upload(to release the pressure of cloud). If there's no such methods, can you give me some advice?
I'm newbee of lora, can you help me?

mttbx, 2019-09-16:
Any help would be graceful

brocaar, 2019-09-19:
Yes, you can use the `DeviceTimeReq` mac-command for this (initiated from the device). Please refer the LoRaWAN specification for more details on this. For batch uploading, you can use the gRPC or REST api.

Next time, please use the forum: https://forum.loraserver.io, it is a better place for general questions. GitHub is better for bugreporting or feature requests.

closed 2019-09-19 by brocaar