brocaar/chirpstack-network-server #432: Introduce the possibility to find entities

eugeneugrevatov, 2019-09-16:
### Is this a bug or a feature request?
Feature

### What did you expect?
I would be very useful to introduce `findAll`/`findByIds` API methods for NetworkServerService

brocaar, 2019-09-19:
Could you explain your use-case and why such feature would be useful?

eugeneugrevatov, 2019-09-20:
@brocaar Hi, description updated.

And one more use-case - fetch all ABP activated devices with activation. It needs to Redis be queried and proprietary format message is parsed. Doesn't look neat either)

So IMO such feature should be presented. Thanx in advance.

brocaar, 2020-10-05:
I'm closing this issue. Currently I'm not planning to support a search feature. The reason for this is that it would require additional indices for the fields you want to search on, which has impact on the performance.

One option that I could see useful is to add key-value attributes to objects, but this would not serve the "fetch all ABP activated devices with activation" use-case.

closed 2020-10-05 by brocaar