brocaar/chirpstack-network-server #435: redis flushall

liangyuanpeng, 2019-09-19:
### Is this a bug or a feature request?

i run redis flushall and i just got some error 
``
 device-session does not exist or invalid fcnt or mic
``
how can i do for devices?

devices awlays re-join

### What did you expect?

### What happened?

device join success and send message success

### What version are your using?

### How can your issue be reproduced?

### Could you share your log output?

brocaar, 2019-09-27:
This question has already been answered on the forum: https://forum.loraserver.io/t/recovering-from-a-redis-flushall/6022.

closed 2019-09-27 by brocaar

stffabi, 2020-03-30:
@brocaar is this really by design that a flush of Redis renders all activations useless? This clearly contradicts to the chirpstack network server documentation here: https://www.chirpstack.io/network-server/install/requirements/#redis-database

> ChirpStack Network Server stores all non-persistent data into a Redis datastore.

This sentence at least for me clearly implies all data stored in the Redis is transient and e.g. don't need to be stored in a backup, because they are all non-persistent.

I think a lot of users won't plan to backup their redis store because of this documentation, so all these users are at risk to loose all activations. If you have widely deployed a lot of sensors out in the world for customers, this could probably result in a little nightmare explaining all customer they have to visit all sites and and reset the sensors to rejoin them.