brocaar/chirpstack-network-server #439: can't update the lora-app-server endpoint after the first request

krasi-georgiev, 2019-09-30:
### Is this a bug or a feature request?
bug

### What version are your using?
compiled from master

```
addrConn.resetTransport failed to create client transport: connection error: desc = "transport: Error while dialing dial tcp 127.0.0.1:8001: getsockopt: connection refused"; Reconnecting to {127.0.0.1:8001 }"

```

Similiar setup like the official the docker setup
https://github.com/brocaar/loraserver-docker/blob/master/docker-compose-env.yml

by mistake I didn't have the  `APPLICATION_SERVER.API.PUBLIC_HOST` env set so the first request to `loraserver` was done with the default `127.0.0.1:8001
`.  Than I fixed the entry to `appserver:8001` but `loraserver` still tries to connect to `127.0.0...` 

I looked at the code and it seems that the first request adds an entry in the postgresql db so every next request only uses an id to fetch that record. 
This make the loraserver use whatever was in the first request so even after changing the lora-app-server endpoint location it still tries to use 127.0.0.1 which was added to the db.

I guess the id needs to be some hash so that if the request is different it creates a new entry.

krasi-georgiev, 2019-10-01:
just experienced a similar race bug when adding a new device.

to replicate:
 - make a join request with a non existing dev eui:40de1b12755d81e3 the request gets cached in the postgres and logs an error that the object doesn't exist.
 - than add the a device with that eui:40de1b12755d81e3
 - make a new request and will get the same error that the object doesn't exist.

```
01.10.19 13:58:49 (+0300)  appserver  INFO[0414] device updated                                ctx_id=e76e559b-e382-4490-8951-45855b742ccb dev_eui=40de1b12755d81e3
01.10.19 13:58:49 (+0300)  appserver  ERRO[0414] get device-activation error: object does not exist  dev_eui=40de1b12755d81e3
01.10.19 13:58:49 (+0300)  appserver  ERRO[0414] finished unary call with code Internal        ctx_id=e76e559b-e382-4490-8951-45855b742ccb error="rpc error: code = Internal desc = get device-activation error: object does not exist" grpc.code=Internal grpc.method=HandleUplinkData grpc.request.deadline="2019-10-01T10:58:50Z" grpc.service=as.ApplicationServerService grpc.start_time="2019-10-01T10:58:49Z" grpc.time_ms=34.341 peer.address="172.18.0.14:42242" span.kind=server system=grpc
```

if I delete the postgres db and create the device first and than send the first request it all works as expected.

krasi-georgiev, 2019-10-25:
ping @brocaar

brocaar, 2019-11-13:
For your first question see: https://github.com/brocaar/chirpstack-network-server/issues/453.

The second one I can't reproduce. I OTAA with a non-existing device which fails as expected with a "not found" error as the device does not exist. Then I create the device and OTAA again, which works as expected.

> the request gets cached in the postgres

I'm not sure to which table you are referring to. I don't believe any requests are cached in PostgreSQL.

krasi-georgiev, 2019-11-13:
ok closing this one and will subscribe to #453 and for the second one will open a new issue if I manage to replicate it again.

closed 2019-11-22 by krasi-georgiev