brocaar/chirpstack-network-server #442: process tree encryption / decryption of project source code

jawadiot, 2019-10-23:
Hi,

   I'm trying to replace the AES algorithm with another encryption algorithm, and for that I try for a long time to list the functions of :
- join request decryption loraserver_join-server(root key).
- join-accept encryption.
- Uplink decryption in loraserver(Network Session Key) and lora-app-server(Application Session Key).
- Downlinkk encryption in loraserver(Network Session Key) and lora-app-server(Application Session Key).

    I'm little confusing with the structure of the project, exemple in  lora-app-server => internal => downlink.go : 

          func EnqueueDownlinkPayload(ctx context.Context, db sqlx.Ext, devEUI lorawan.EUI64, confirmed bool, fPort uint8, data []byte) (uint32, error) {
              .
              .
              .
          // encrypt payload
	       b, err := lorawan.EncryptFRMPayload(da.AppSKey, false, da.DevAddr, resp.FCnt, data)
	        if err != nil {
		        return 0, errors.Wrap(err, "encrypt frmpayload error")
	      }

i can't find reference of : 
**` EncryptFRMPayload `**  where that supose to refer to some aes cipher function

  If some one is very accustomedwith with the source code of the project, could you help me to find the good functions of encryption and decryption? i know that the final function use the package "crypto/aes", but i need to fund the functions that call them, thank you.

Best regard

brocaar, 2019-10-23:
It is part of the `lorawan` package (see the import statement). Please see: https://github.com/brocaar/lorawan.

closed 2019-10-23 by brocaar

jawadiot, 2019-10-23:
thanks,

   When we build the project, the **lorawan package** is automatically generated in 
 **pkg/mod/github.com/brocaar** .
   Which the file that downloads and decompresses this package in the beginin of the build? and can I just put the **lorawan package** in **src/github.com/brocaar** and disabled which is in **pkg/mod/github.com/brocaar** ?

jawadiot, 2019-10-24:
Thanks a lot !