brocaar/chirpstack-network-server #443: Lora-app-server  not running

varunsavita, 2019-10-24:
### Is this a bug or a feature request?

### What did you expect?

### What happened?

### What version are your using?

### How can your issue be reproduced?

### Could you share your log output?

varunsavita, 2019-10-24:
FATA[0000] error loading config file                     config=/etc/lora-app-server/lora-app-server.toml error="open /etc/lora-app-server/lora-app-server.toml: permission denied"

wolftrax5, 2019-10-24:
I think the problem is not access to config, for permissions, try run it with sudo

varunsavita, 2019-10-28:
ubuntu@ip-172-31-25-243:~$ lora-app-server --config /etc/lora-app-server/lora-app-server.toml
FATA[0000] error loading config file                     config=/etc/lora-app-server/lora-app-server.toml error="open /etc/lora                       -app-server/lora-app-server.toml: permission denied"
ubuntu@ip-172-31-25-243:~$ sudo lora-app-server --config /etc/lora-app-server/lora-app-server.toml
INFO[0000] starting LoRa App Server                      docs="https://www.loraserver.io/" version=3.4.0-test.2
INFO[0000] storage: setting up storage package
INFO[0000] storage: setup metrics
INFO[0000] storage: setting up Redis pool
INFO[0000] storage: connecting to PostgreSQL database
INFO[0000] storage: applying PostgreSQL data migrations
FATA[0000] setup storage error: storage: applying PostgreSQL data migrations error: pq: type "hstore" does not exist handling 0                       041_device_variables.sql
ubuntu@ip-172-31-25-243:~$

varunsavita, 2019-10-28:
loraserver_as=> create extension hstore;
ERROR:  permission denied to create extension "hstore"
HINT:  Must be superuser to create this extension.

varunsavita, 2019-10-28:
i created "create extension hstore;"
now this error came.

ubuntu@ip-172-31-25-243:~$ sudo lora-app-server --config /etc/lora-app-server/lora-app-server.toml
INFO[0000] starting LoRa App Server                      docs="https://www.loraserver.io/" version=3.4.0-test.2
INFO[0000] storage: setting up storage package
INFO[0000] storage: setup metrics
INFO[0000] storage: setting up Redis pool
INFO[0000] storage: connecting to PostgreSQL database
INFO[0000] storage: applying PostgreSQL data migrations
INFO[0000] storage: PostgreSQL data migrations applied   count=0
INFO[0000] integration/mqtt: TLS config is empty
INFO[0000] integration/mqtt: connecting to mqtt broker   server="tcp://localhost:1883"
INFO[0000] api/as: starting application-server api       bind="0.0.0.0:8001" ca_cert= tls_cert= tls_key=
FATA[0000] setup api error: setup application-server api error: start application-server api listener error: listen tcp 0.0.0.0:8001: bind: address already in use
ubuntu@ip-172-31-25-243:~$

brocaar, 2019-11-13:
I'm closing this issue as this is related to configuration, not a bug or feature request. I think the error explains well what the issue is:

> start application-server api listener error: listen tcp 0.0.0.0:8001: bind: address already in use

closed 2019-11-13 by brocaar