brocaar/chirpstack-network-server #459: add network server from lora-app

bashhole, 2019-12-12:
Hello Orne, please tell me? how add second loraserver from lora-app interface)
I create new config from another band and run loraserver -c /etc/loraserver/second.toml
Interface lora app show me error message and isee in console error
`INFO[0107] finished unary call with code AlreadyExists   error="rpc error: code = AlreadyExists desc = object already exists" grpc.code=AlreadyExists grpc.method=CreateRoutingProfile grpc.service=ns.NetworkServerService grpc.start_time="2019-12-12T11:27:16Z" grpc.time_ms=1.537 peer.address="[::1]:39610" span.kind=server system=grpc`
What the problem? i think second loraserver  must have new database?

entrehuihui, 2020-01-17:
> 你好奥恩，请告诉我？如何从lora-app界面添加第二个loraserver）
> 我从另一个频段创建新配置并运行loraserver -c /etc/loraserver/second.toml
> 接口lora应用向我显示错误消息，并在控制台错误中显示isee是
> `INFO[0107] finished unary call with code AlreadyExists error="rpc error: code = AlreadyExists desc = object already exists" grpc.code=AlreadyExists grpc.method=CreateRoutingProfile grpc.service=ns.NetworkServerService grpc.start_time="2019-12-12T11:27:16Z" grpc.time_ms=1.537 peer.address="[::1]:39610" span.kind=server system=grpc`
> 什么问题？我认为第二个loraserver必须具有新数据库？

The second server must also have a new database

brocaar, 2020-01-21:
The above is correct and should solve your issue.

closed 2020-01-21 by brocaar