brocaar/chirpstack-network-server #471: Error="get device-session error: invalid MIC"

OniremE, 2020-03-05:
Hi, i have this error when i send some data to the network server:
mar 05 12:23:16 onirem chirpstack-network-server[970]: time="2020-03-05T12:23:16-05:00" level=info msg="finished client unary call" ctx_id=f4a36a55-1dce-4642-a959-c8d5e5419a2a grpc.code=OK grpc.ctx_id=29205a7a-c5c7-4ae7-b042-02a80ea2cd19 grpc.duration=15.35368ms grpc.method=HandleError grpc.service=as.ApplicationServerService span.kind=client system=grpc
mar 05 12:23:16 onirem chirpstack-network-server[970]: time="2020-03-05T12:23:16-05:00" level=error msg="uplink: processing uplink frame error" ctx_id=f4a36a55-1dce-4642-a959-c8d5e5419a2a error="get device-session error: invalid MIC"

OniremE, 2020-03-06:
the problem is solved. i'll close.

closed 2020-03-06 by OniremE

mo-norant, 2021-01-21:
could you elaborate please