brocaar/chirpstack-network-server #472: Network Server otaa object does not exist

milakhs, 2020-03-07:
### Is this a bug or a feature request?
bug?
### What did you expect?
The network server to activate a lora sensor over the air and to send a downlink command to the commands pub/sub topic.
### What happened?
Set up Chirpstack network server and application server following the "Quickstart Google Cloud Platform" twice whilst going through ever step meticulously. Every log check seem to output what the examples in the guide proscribe. Yet when adding the device eui in the application server, the network server doesn't seem to send a downlink command to gcp pub/sub and also outputs the following error => error="rpc error: code = NotFound desc = object does not exist" grpc.code=NotFound. When checking the as923-gateway-commands pub/sub topic no messages have been recorded. 
### What version are your using?
Network Server 3.8.0 and Application Server 3.9.0
### How can your issue be reproduced?

### Could you share your log output?
Network Server:
Mar 05 23:09:24 lora-server-instance chirpstack-application-server[10609]: time="2020-03-05T23:09:24Z" level=info msg="finished client unary call" ctx_id=189c14c7-2911-4f7d-8c15-84c0b9e80f4e grpc.code=OK grpc.ctx_id=e057b062-9a0f-4d89-8761-c493b33b4f9a grpc.duration=2.022226ms grpc.method=GetDeviceProfile grpc.service=ns.NetworkServerService span.kind=client system=grpc
Mar 05 23:09:24 lora-server-instance chirpstack-application-server[10609]: time="2020-03-05T23:09:24Z" level=info msg="finished unary call with code OK" ctx_id=189c14c7-2911-4f7d-8c15-84c0b9e80f4e grpc.code=OK grpc.method=Get grpc.service=api.DeviceProfileService grpc.start_time="2020-03-05T23:09:24Z" grpc.time_ms=7.28 peer.address="127.0.0.1:53696" span.kind=server system=grpc
Mar 05 23:09:24 lora-server-instance chirpstack-application-server[10609]: time="2020-03-05T23:09:24Z" level=info msg="finished unary call with code NotFound" ctx_id=0335ebbe-db90-4ee9-9b7a-f744defaf1a6 error="rpc error: code = NotFound desc = object does not exist" grpc.code=NotFound grpc.method=GetKeys grpc.service=api.DeviceService grpc.start_time="2020-03-05T23:09:24Z" grpc.time_ms=3.315 peer.address="127.0.0.1:53686" span.kind=server system=grpc
Mar 05 23:09:34 lora-server-instance chirpstack-application-server[10609]: time="2020-03-05T23:09:34Z" level=info msg="device-keys created" ctx_id=d0e1d8a5-070a-4bf0-bada-0cf84624ef11 dev_eui=70b3d57ed0015c38

Application server:
Mar 05 23:09:37 lora-server-instance chirpstack-network-server[10623]: time="2020-03-05T23:09:37Z" level=info msg="finished client unary call" ctx_id=3530f84a-7cc1-4ae4-af8c-3e0965ae50fa grpc.code=OK grpc.ctx_id=b5eb60b4-f5cb-40cf-aa0d-442665d070f0 grpc.duration=6.69165ms grpc.method=HandleGatewayStats grpc.service=as.ApplicationServerService span.kind=client system=grpc
Mar 05 23:09:39 lora-server-instance chirpstack-network-server[10623]: time="2020-03-05T23:09:39Z" level=info msg="finished unary call with code OK" ctx_id=33271747-0958-4f43-8039-5ba7ea613e7e grpc.code=OK grpc.method=GetDevice grpc.service=ns.NetworkServerService grpc.start_time="2020-03-05T23:09:39Z" grpc.time_ms=1.023 peer.address="127.0.0.1:52150" span.kind=server system=grpc
Mar 05 23:09:39 lora-server-instance chirpstack-network-server[10623]: time="2020-03-05T23:09:39Z" level=info msg="finished unary call with code OK" ctx_id=0869cfbf-1920-4b9d-b746-22955dd4d478 grpc.code=OK grpc.method=GetDeviceProfile grpc.service=ns.NetworkServerService grpc.start_time="2020-03-05T23:09:39Z" grpc.time_ms=1.197 peer.address="127.0.0.1:52150" span.kind=server system=grpc
Mar 05 23:09:39 lora-server-instance chirpstack-network-server[10623]: time="2020-03-05T23:09:39Z" level=info msg="finished unary call with code NotFound" ctx_id=860959a0-481d-4d1f-9c06-6b25a2954e35 error="rpc error: code = NotFound desc = object does not exist" grpc.code=NotFound grpc.method=GetDeviceActivation grpc.service=ns.NetworkServerService grpc.start_time="2020-03-05T23:09:39Z" grpc.time_ms=0.475 peer.address="127.0.0.1:52150" span.kind=server system=grpc
Mar 05 23:10:07 lora-server-instance chirpstack-network-server[10623]: time="2020-03-05T23:10:07Z" level=info msg="gateway/gcp_pub_sub: stats event received" gateway_id=00800000a000279e stats_id=bc5bd7fa-12fd-4138-a470-d68fc9d15ac0
Mar 05 23:10:07 lora-server-instance chirpstack-network-server[10623]: time="2020-03-05T23:10:07Z" level=info msg="gateway updated" ctx_id=bc5bd7fa-12fd-4138-a470-d68fc9d15ac0 gateway_id=00800000a000279e

brocaar, 2020-03-31:
> error="rpc error: code = NotFound desc = object does not exist" grpc.code=NotFound grpc.method=GetKeys

This is usually an indication that you made a typo in the DevEUI. Or you did not enter the AppKey so no key can be found the the DevEUI trying to join the network.

closed 2020-03-31 by brocaar

milakhs, 2020-04-01:
Hi brocaar, thanks for responding. As far as the DevEUI and AppKey, I have triple checked those. The lora device that has been used for this test is an Elsys ERS device that allows to check those parameters on the device, so I'm pretty sure they have been put in correctly. Also this device can be added to TTN with those keys with no issues. Could there be another issue that causes this message?