brocaar/chirpstack-network-server #476: Class C Downlink Sent On Wrong Channel

encojosh, 2020-03-25:
### Is this a bug or a feature request?
Bug
### What did you expect?
All class C downlinks to be sent on the frequency set in config file via the rx2_frequency parameter.
### What happened?
Class C downlinks sent on default band 923300000 instead of configured 925100000.
### What version are your using?
3.8.1
### How can your issue be reproduced?
Device sends a confirmed uplink.  Using MQTT, we listen for the confirmed uplink and immediately schedule a unconfirmed downlink.  The empty ACK packet is sent in the RX1 window and the queued downlink is sent afterwards on 923300000 instead of 925100000.  All multicasts are sent on the 925100000, but unicast Class C downlinks seem to be stuck on 923300000.  
### Could you share your log output?

encojosh, 2020-03-25:
[0004a30b0020d7fa.log](https://github.com/brocaar/chirpstack-network-server/files/4382949/0004a30b0020d7fa.log)

encojosh, 2020-03-25:
[chirpstack-network-server.zip](https://github.com/brocaar/chirpstack-network-server/files/4383130/chirpstack-network-server.zip)

config file

encojosh, 2020-03-26:
If I delay the downlink by 2.5s (after the RX1 window closes), the downlink is still sent on the default 923300000 instead of 925100000.

brocaar, 2020-03-31:
When your configuration is different than the default, the NS will send a mac-command to adjust these settings on the device. As long the device does not ACK these new settings, the NS will continue to use the defaults.

See:
https://github.com/brocaar/chirpstack-network-server/blob/master/internal/maccommand/rx_param_setup.go#L15

Multicast groups have their own settings as these are configured out of band.

encojosh, 2020-03-31:
Yes, this is an issue with our specific use case.  We were forced to turn off ADR and all MAC commands due to some bugs in our RN2903 FW, so we decided to distribute the downlink channel and data rate in the application layer.  However, chirpstack was unaware of the application layer changes, so Class C downlinks are still sent on the default downlink channel.  This can be modified using gRPC.  Also, our Class C scheduler is set to 10 ms which wasn't allowing enough time to schedule a Class A downlink via the API.

closed 2020-03-31 by encojosh