brocaar/chirpstack-network-server #481: Cannot get latest version: module contains a go.mod file, so module path should be github.com/brocaar/chirpstack-network-server/v3

KateGo520, 2020-05-06:
## Background
The github.com/brocaar/chirpstack-network-server uses [Go modules](https://github.com/golang/go/wiki/Modules) and the current release stream version is **v3**. It must comply with the specification of "Releasing Modules for v2 or higher" available in the Modules documentation and enforced since the most recent versions of Go. Quoting the specification:
> A package that has opted in to modules _must_ include the major version in the import path to import any v2+ modules
> To preserve import compatibility, the go command requires that modules with major version v2 or later use a module path with that major version as the final element. For example, version v2.0.0 of _example.com/m_ must instead use module path _example.com/m/v2_.
https://github.com/golang/go/wiki/Modules#releasing-modules-v2-or-higher

Using `go get` (under go1.13 and go1.14) does not work for anything later than github.com/brocaar/chirpstack-network-server@v2.5.0, which was the last version to not have modules.
I consider this a pretty **high severity** issue since not doing so **prevents any other Go project also using Go Modules** to properly require ` brocaar/chirpstack-network-server ` on versions higher than v2.5.0 as a dependency, resulting in errors like:
```
invalid version: module contains a go.mod file, so major version must be compatible: should be v0 or v1, not v3
```

SO anyone using Go modules will not be able to easily use any newer version of brocaar/chirpstack-network-server

## Steps to Reproduce
1. Under a project making use of Go modules, run `go get` targeting any version >= v2.6.0 of the brocaar/chirpstack-network-server:

```
$ go get github.com/brocaar/chirpstack-network-server@v3.9.0
go: finding github.com/brocaar/chirpstack-network-server v3.9.0
go: finding github.com/brocaar/chirpstack-network-server v3.9.0
go get github.com/brocaar/chirpstack-network-server@v3.9.0: github.com/ brocaar/chirpstack-network-server@v3.9.0: invalid version: module contains a go.mod file, so major version must be compatible: should be v0 or v1, not v3
```

## Solution
### Kill the go.mod files.
This would push them back to not being managed by Go modules (instead of incorrectly using Go modules). Importing specific versions would work, but they'd receive the `+incompatible` prefix and would need to be manually upgraded.

Probably best option short-term -- better to not use Go modules than use it incorrectly.

Also, best option for older branches/versions that aren't maintained.

### Make breaking changes
Patch the `go.mod` file to declare the `module` as `github.com/brocaar/chirpstack-network-server/v3` as [per the specs](https://github.com/golang/go/wiki/Modules#semantic-import-versioning). AND adjust all internal imports.

This would probably only be acceptable on the v3 line, since it'd break any consumers.
Two methods can be use:
* a new branch in the same repo
* use the folders structure on mainline branch (see Major Subdirectory structure on the [Modules page](https://github.com/golang/go/wiki/Modules#releasing-modules-v2-or-higher)).

Then development would take place in the new branch/repository, with the existing only updated with tags and/or with new things to forward.

### Major version bump / repository change
Leave `v3` as a dead version and bump to v4 with the path changes.

### References
* https://github.com/golang/go/wiki/Modules#semantic-import-versioning
* https://golang.org/cmd/go/#hdr-Module_compatibility_and_semantic_versioning
* https://github.com/golang/go/wiki/Modules#releasing-modules-v2-or-higher

KateGo520, 2020-05-06:
@brocaar

brocaar, 2020-05-06:
`github.com/brocaar/chirpstack-network-server` is not a library and should not be imported by other projects. Actually everything is under `internal/` so there is not so much to import.

If you are interested in using the API endpoints, you can import `https://github.com/brocaar/chirpstack-api/go/v3`.

closed 2020-05-06 by brocaar