brocaar/chirpstack-network-server #483: Multicast-group devices not received normal downlink from the server

sagar-patel-sls, 2020-05-25:
- [x] The issue is present in the latest release.
- [x] I have searched the [issues](https://github.com/brocaar/chirpstack-network-server) of this repository and believe that this is not a duplicate.

[Thread referencing this request](https://forum.chirpstack.io/t/max-number-of-devices-that-can-be-associated-with-a-multicast-group/8193/5)

## What happened?
We have created a multicast-group and added 5 devices in the group.
We have sent multicast downlink on that group and successfully received multicast downlink on all the devices.

But when we have sent normal downlink on the single device, the device has not received any downlink from the server.

**Multicast-group joined device not received normal downlink (received only multicast downlink)**
## What did you expect?
We expect the device should receive a multicast downlink as well as a normal downlink from the server.
## Steps to reproduce this issue

Steps:

1. Create Multicast Group
2. Add 4-5 devices in the multicast-group
3. Send multicast downlink using Swagger API (No GUI implemented in chirpstack-application-server)
4. After the multicast downlink successfully received in devices
5. Send normal downlink on the single device from chirpstack-application-server web-interface

## Your Environment

| Component           | Version |
| --------------------| ------- |
| Application Server  | v3.10.0 |
| Network Server      |  v3.9.0 |
| Gateway Bridge      | v3.8.0 |
| Chirpstack API      |         |
| Geolocation         |         |
| Concentratord       |         |

brocaar, 2020-05-25:
Are these Class-A or Class-C devices?

sagar-patel-sls, 2020-05-25:
For Class-C devices

brocaar, 2020-05-25:
Then it is still not clear from the issue if the downlink is sent but not received or the downlink remains in the queue. Could you provide:

1. The confirmation that the downlinks remain in the downlink queue -or-
2. Logs showing if 1. is not the case, these downlinks are emitted

sagar-patel-sls, 2020-05-25:
1. Multicast downlink was successfully received from the server to devices
2. Normal downlink not received to any multicast-group joined device

The downlink is sent but not received

brocaar, 2020-05-25:
> Normal downlink not received to any multicast-group joined device

Have you tried to send a downlink to the same type of device that is not part of a multicast-group?

> The downlink is sent but not received

Any logs for debugging would be helpful. It would be good to confirm if the downlink frequency, data-rate etc.. are correct for the unicast Class-C downlink, as would any gateway log confirming that the downlink is actually emitted by the gateway.

sagar-patel-sls, 2020-05-25:
> Have you tried to send a downlink to the same type of device that is not part of a multicast-group?

Yes, when we delete the multicast group and keys from the devices it will work fine with normal downlink

> Any logs for debugging would be helpful.

**Multicast Group Configuration**
DR: 5
Frequency: 865985000  (IN865 ISM Band)
GroupType: Class C

brocaar, 2020-05-25:
> Yes, when we delete the multicast group and keys from the devices it will work fine with normal downlink

If after making a change to the device **only** without any change in ChirpStack the downlink works, then there is no issue with ChirpStack.

Please note that when the unicast Class-C downlink is sent on a different freq. or data-rate than the downlink multicast, this is by definition not going to work as a device can not listen on two frequencies at a single time.

sagar-patel-sls, 2020-05-25:
@brocaar 

we are not using Local Multicast we have developed Remote multicasting in `chirpstack-application-server`

we have modified `chirpstack-application-server` source-code for Remote Multicast features based on below specification and commands

[https://lora-alliance.org/sites/default/files/2018-09/remote_multicast_setup_v1.0.0.pdf](https://lora-alliance.org/sites/default/files/2018-09/remote_multicast_setup_v1.0.0.pdf)

brocaar, 2020-05-25:
I can't support you when you are using your own modified version of the Chirpstack Application Server...

But still this could explain your issue:

Please note that when the unicast Class-C downlink is sent on a different freq. or data-rate than the downlink multicast, this is by definition not going to work as a device can not listen on two frequencies at a single time.

sagar-patel-sls, 2020-05-26:
Thank You @brocaar 

> Please note that when the unicast Class-C downlink is sent on a different freq. or data-rate than the downlink multicast, this is by definition not going to work as a device can not listen on two frequencies at a single time.

If we use `chirpstack-application-server` **Local Multicast** then normal downlink also not working?

brocaar, 2020-05-26:
If I understand your situation correctly, then you have one Class-C device which is also part of a Class-C multicast session. In case of Class-C, a device is always listening **on a single frequency and data-rata**. So if the frequency & data-rate of the multicast-group is A and the frequency & data-rate of the unicast Class-C is B, then your device can only receive the multicast **or** the unicast.

If the freq & data-rate are the same for multicast and unicast, then this might work as the device could detect if it is an unicast or multicast DevAddr, but this depends on your device firmware.

sagar-patel-sls, 2020-05-26:
> If the freq & data-rate are the same for multicast and unicast, then this might work as the device could detect if it is a unicast or multicast DevAddr, but this depends on your device firmware.

The freq & data-rate are the same for multicast & unicast is working for us.

Thank you @brocaar

closed 2020-06-12 by brocaar