brocaar/chirpstack-network-server #485: High CPU usage

momobel, 2020-06-11:
Hi,

Our goal is to run Chirpstack network server on embedded device with relatively low power (~500MHz ARMv7 single core CPU, 256MB RAM, 256MB NAND).
We are benchmarking Chirpstack and https://github.com/gotthardp/lorawan-server to get a better idea.

Thanks for your help!

- [ ] The issue is present in the latest release.
- [x] The issue is present on version 3.6 (but nothing related to performance seem included in latest releases)
- [x] I have searched the [issues](https://github.com/brocaar/chirpstack-network-server) of this repository and believe that this is not a duplicate.

## What happened?

We noticed that Chirpstack network server (without application server) seems to consume more CPU on average (~20 to ~30%) compared to lorawan-server as can be seen in the screenshot below.
On the other hand, Chirpstack seems more consistent since lorawan-server has some CPU spikes (up to ~30%) compared to its cruise CPU usage.

full 890s

[image]

zoom of 60s around 400s:

[image]

Of course this may be because of various factors and we are keen on understanding what is using the CPU.
We are curious about what may explain this difference, do you have any ideas of the cause?
Is there any configuration that we can tune to achieve lower CPU usage?
Similarly would there be any low-hanging fruit that we could change on Chirpstack to shave some CPU usage?

## What did you expect?

Nothing in particular, we are just investigating and analyzing.

## Steps to reproduce this issue

- Run network server
- Add 15 ABP devices
- Simulate 1 uplink frame per device every 6 seconds through a packet forwarder simulator
- Measure CPU usage with `pidstat`

## Could you share your log output?

N/A

## Your Environment

ARMv7

| Component           | Version |
| --------------------| ------- |
| Application Server  | v3.6  |
| Network Server      |  v3.6    |
| Gateway Bridge      |  v3.7   |

brocaar, 2020-06-12:
This could be an interesting source when you would like to dive into CPU profiling: https://blog.golang.org/pprof. With this information, you can track how much CPU time is spent per function / method.

momobel, 2020-06-15:
Thanks for your feedback.

We have run pprof and it seems the main CPU consumers are syscalls, mutex, and waits.

```      flat  flat%   sum%        cum   cum%
     9.62s 21.75% 21.75%     10.63s 24.04%  syscall.Syscall
     4.71s 10.65% 32.41%      4.71s 10.65%  runtime.futex
     3.83s  8.66% 41.07%      3.83s  8.66%  runtime.usleep
     3.05s  6.90% 47.96%      3.05s  6.90%  runtime.epollwait
     0.91s  2.06% 50.02%      3.01s  6.81%  runtime.mallocgc
     0.51s  1.15% 51.18%      7.92s 17.91%  runtime.findrunnable
     0.51s  1.15% 52.33%      0.51s  1.15%  runtime.madvise
     0.50s  1.13% 53.46%      0.50s  1.13%  runtime.memmove
     0.48s  1.09% 54.55%      0.80s  1.81%  runtime.scanobject
     0.46s  1.04% 55.59%      0.55s  1.24%  runtime.nanotime (inline)
     0.40s   0.9% 56.49%      0.73s  1.65%  runtime.heapBitsSetType
     0.37s  0.84% 57.33%      0.37s  0.84%  runtime.memclrNoHeapPointers
     0.35s  0.79% 58.12%      0.35s  0.79%  runtime.nextFreeFast (inline)
     0.35s  0.79% 58.91%      0.35s  0.79%  runtime.write1
     0.31s   0.7% 59.61%      0.35s  0.79%  runtime.lock
     0.29s  0.66% 60.27%      0.30s  0.68%  runtime.(*itabTableType).find
     0.29s  0.66% 60.92%      1.83s  4.14%  runtime.newobject
     0.22s   0.5% 61.42%      0.34s  0.77%  reflect.StructTag.Lookup
     0.21s  0.47% 61.90%      0.32s  0.72%  strconv.appendEscapedRune
     0.20s  0.45% 62.35%      2.40s  5.43%  github.com/sirupsen/logrus.(*TextFormatter).Format```

ABalanuta, 2020-06-29:
I believe I have a similar issue.
I am running the Network server inside a Docker container on a RPI4 and the average load is always around 2.5 . After enabling the debug flag There is a loop happening when the downlink device-queue scheduler is initiated.
Here is a fragment of the logs that is looped ~1000/sec:
```
time="2020-06-29T20:26:38Z" level=info msg="starting downlink device-queue scheduler"
time="2020-06-29T20:26:38Z" level=info msg="starting multicast scheduler"
time="2020-06-29T20:26:38Z" level=debug msg="running class-b / class-c scheduler batch" ctx_id=196c2106-fc18-4439-b822-ac312e888833
time="2020-06-29T20:26:38Z" level=debug msg="running multicast scheduler batch" ctx_id=284f0033-4fc7-4373-8e5e-dc6059dd59ab
time="2020-06-29T20:26:38Z" level=debug msg="sql query executed" args="[100 354860h26m56.392225352s 2020-06-29 20:26:38.39023187 +0000 UTC m=+2.199008984]" duration=10.155149ms query="\n        select\n            d.*\n        from\n            device d\n        where\n\t\t\td.mode in ('B', 'C')\n            -- we want devices with queue items\n            and exists (\n                select\n                    1\n                from\n                    device_queue dq\n                where\n                    dq.dev_eui = d.dev_eui\n                    and (\n\t\t\t\t\t\td.mode = 'C'\n                    \tor (\n\t\t\t\t\t\t\td.mode = 'B'\n                    \t\tand dq.emit_at_time_since_gps_epoch <= $2\n                    \t)\n                    )\n            )\n            -- we don't want device with pending queue items that did not yet\n            -- timeout\n            and not exists (\n                select\n                    1\n                from\n                    device_queue dq\n                where\n                    dq.dev_eui = d.dev_eui\n                    and is_pending = true\n                    and dq.timeout_after > $3 \n            )\n        order by\n            d.dev_eui\n        limit $1\n        for update of d skip locked"
time="2020-06-29T20:26:38Z" level=debug msg="sql query executed" args="[100 2020-06-29 20:26:38.398065026 +0000 UTC m=+2.206842177]" duration=4.612228ms query="\n\t\tselect\n\t\t\t*\n\t\tfrom\n\t\t\tmulticast_queue\n\t\twhere\n\t\t\tschedule_at <= $2\n\t\torder by\n\t\t\tid\n\t\tlimit $1\n\t\tfor update skip locked\n\t"
time="2020-06-29T20:26:38Z" level=debug msg="running class-b / class-c scheduler batch" ctx_id=6e9e22e7-bd68-4607-aaaf-df51b3875cf7
time="2020-06-29T20:26:38Z" level=debug msg="running multicast scheduler batch" ctx_id=6223a7a6-65c2-4de7-a153-80da03ad1088
time="2020-06-29T20:26:38Z" level=debug msg="sql query executed" args="[100 354860h26m56.405791742s 2020-06-29 20:26:38.403796649 +0000 UTC m=+2.212573762]" duration=2.491343ms query="\n        select\n            d.*\n        from\n            device d\n        where\n\t\t\td.mode in ('B', 'C')\n            -- we want devices with queue items\n            and exists (\n                select\n                    1\n                from\n                    device_queue dq\n                where\n                    dq.dev_eui = d.dev_eui\n                    and (\n\t\t\t\t\t\td.mode = 'C'\n                    \tor (\n\t\t\t\t\t\t\td.mode = 'B'\n                    \t\tand dq.emit_at_time_since_gps_epoch <= $2\n                    \t)\n                    )\n            )\n            -- we don't want device with pending queue items that did not yet\n            -- timeout\n            and not exists (\n                select\n                    1\n                from\n                    device_queue dq\n                where\n                    dq.dev_eui = d.dev_eui\n                    and is_pending = true\n                    and dq.timeout_after > $3 \n            )\n        order by\n            d.dev_eui\n        limit $1\n        for update of d skip locked"
time="2020-06-29T20:26:38Z" level=debug msg="sql query executed" args="[100 2020-06-29 20:26:38.405343206 +0000 UTC m=+2.214120338]" duration=1.281023ms query="\n\t\tselect\n\t\t\t*\n\t\tfrom\n\t\t\tmulticast_queue\n\t\twhere\n\t\t\tschedule_at <= $2\n\t\torder by\n\t\t\tid\n\t\tlimit $1\n\t\tfor update skip locked\n\t"
time="2020-06-29T20:26:38Z" level=debug msg="running class-b / class-c scheduler batch" ctx_id=d12ce13c-24e7-49af-898f-8859f477fe5f
time="2020-06-29T20:26:38Z" level=debug msg="running multicast scheduler batch" ctx_id=4c467af5-d4ae-4ab0-88fc-81b7a8d254ba
time="2020-06-29T20:26:38Z" level=debug msg="sql query executed" args="[100 2020-06-29 20:26:38.408846002 +0000 UTC m=+2.217623171]" duration=1.200617ms query="\n\t\tselect\n\t\t\t*\n\t\tfrom\n\t\t\tmulticast_queue\n\t\twhere\n\t\t\tschedule_at <= $2\n\t\torder by\n\t\t\tid\n\t\tlimit $1\n\t\tfor update skip locked\n\t"
time="2020-06-29T20:26:38Z" level=debug msg="sql query executed" args="[100 354860h26m56.410748262s 2020-06-29 20:26:38.408754651 +0000 UTC m=+2.217531764]" duration=2.166162ms query="\n        select\n            d.*\n        from\n            device d\n        where\n\t\t\td.mode in ('B', 'C')\n            -- we want devices with queue items\n            and exists (\n                select\n                    1\n                from\n                    device_queue dq\n                where\n                    dq.dev_eui = d.dev_eui\n                    and (\n\t\t\t\t\t\td.mode = 'C'\n                    \tor (\n\t\t\t\t\t\t\td.mode = 'B'\n                    \t\tand dq.emit_at_time_sin [... cut ...] select\n                    1\n                from\n                    device_queue dq\n                where\n                    dq.dev_eui = d.dev_eui\n                    and is_pending = true\n                    and dq.timeout_after > $3 \n            )\n        order by\n            d.dev_eui\n        limit $1\n        for update of d skip locked"
time="2020-06-29T20:26:38Z" level=debug msg="running multicast scheduler batch" ctx_id=3bd6125d-b8f7-4058-8df2-a609bcc12c4b
... x1000/sec
```

brocaar, 2020-10-05:
I'm closing this issue. While I would be interested in performance optimizations, this issue is too generic to start working on. Also the two solutions that are compared are quite different (also in terms of features), so CPU usage difference is expected. My previous comment does provide some information on how to profile Go applications. If you find a function which should be optimized, then please create a new issue :)

@ABalanuta: if you see ~ 1000/sec logs for the scheduler, then this could be a config error. I can not reproduce the same behavior.

closed 2020-10-05 by brocaar